slic3r/Slic3r #2621: Bridges with holes/saggings

Lord2Vader, 2015-02-02:
Hi,

most of the time i print mechanical stuff, which has many overhangs or holes with saggings/sink (sorry iam german i dont know the word for it).

Well. The picture tells it. There is a hole in the middle of a sink and slic3r does only form the bridges in one direction/dimension, so only the part which travels beside the hole gets bridged correctly.
Maybe you would say print it upside down but in many cases this is not possible due to overhang/shrinkingproblems

[image]

Also when perimeters first is activated (or NOT infill first) then the inner perimeter of the hole gets printed in mid air.

Here another pic:
[image]

alexrj, 2015-02-02:
Well… you need support material.  No other solutions :-)

(or you can put a membrane in your model, that you can easily drill out after printing)

nophead, 2015-02-02:
Or you might be able to use the technique described here:
http://hydraraptor.blogspot.co.uk/2014/03/buried-nuts-and-hanging-holes.html.
Not sure if it works with Slic3r.

alexrj, 2015-02-02:
Also, a related idea was discussed in #254

Lord2Vader, 2015-02-02:
Well how about bridging besides the hole and the overlapp-bridging on the same layer in the other direction? How could that not be possible?

lordofhyphens, 2017-03-19:
You still need to place that perimeter and attach it to something that isn't in free space (aka support).

closed 2017-03-19 by lordofhyphens