slic3r/Slic3r #2624: 1.2.6-exp on windows crash on loading a STL

gege2b, 2015-02-03:
Hi all
Win7 x64
Slic3r 1.2.6-exp x64

Every time a load a STL, slic3r crash, but only if I have draged and zoomed the (empty) 3D view before.
If I don't move the view before loading a STL, it does not crash

I was able to reproduce this every time, using many different STL, but not with the default config.

So here is my config file : https://dl.dropboxusercontent.com/u/34934879/zoom_drag_crash_config.ini (snipped the octoprint stuff)

and if you want, one of the STL  that caused a crash : https://dl.dropboxusercontent.com/u/34934879/darthVader_cleaned_fixed.stl

Kopfgeldjaeger, 2015-02-04:
Similar problem. Win 8.1 x64, Slic3r 1.2.6
Crashes when loading any STL file at the "Infilling  layers..." step. Doesn't matter whether 3D or 2D view is clicked.
config.ini: http://nopaste.linux-dev.org/?429221

I updated from Slic3r 1.1.7 just before.

gege2b, 2015-02-04:
well, I don't konw if and how this can be related, but I got a crash everytime and ONLY if I dragged/zoomed the 3D view before loading the STL

If I don't do that, Slic3r didn't crash

Haasoperator, 2015-02-08:
I am having the same problem, it even happens when I try to use the 2D plater view and don't even touch the 3D plater. As soon as I load STL's on the plater, it immediately, without prompting, begins fiddling with triangular mesh, generating perimeters, infill, etc., then shuts down. THis is REALLY frustrating, as I want to go use the new XY adjust feature as well as the 3D plating view and other new features, but it's impossible.

quintesse, 2015-02-11:
Have this as well. And can confirm that setting the print/filamente/printer settings to ` - default - `, then exiting and restarting fixes the problem (as long as you do this every time).

Haasoperator, 2015-02-11:
But what if you don't want to use the default settings?

Kopfgeldjaeger, 2015-02-12:
Haasoperator wrote:
> But what if you don't want to use the default settings?

You just have to *start* Slic3r with default settings. When started, you can change the settings to what you want. Before closing Slic3r, you should then again revert to default settings.

axs-gentoo, 2015-02-24:
I have an issue similar to this, but i can start with default settings and still have the crash.  In my case, I'm running windows in a virtualbox vm, and i was able to keep it from crashing by setting it to single-threaded mode and turning off background processing.  This is on a 32bit windows, if it matters.

Nas-r, 2015-05-10:
I'm having the same issue. Starting up with default settings and changing after that seems to fix it.
- windows 8 64bit

alexrj, 2015-05-24:
Can you try the newly released 1.2.7 and check whether it fixes this issue? Thank you!

quintesse, 2015-05-24:
@alexrj I just did and it seems to be fixed on my system at least, thanks!

alexrj, 2015-05-24:
@quintesse, thank you for testing!  Crossing fingers and waiting for feedback from other people who reported the problem here.

gege2b, 2015-05-25:
seems OK here too!

Nas-r, 2015-05-25:
Issue is resolved for me as well. Awesome!

closed 2015-06-01 by alexrj