slic3r/Slic3r #2625: Can't load OpenGL.so for module OpenGL in 1.2.5 and 1.2.6 (x86) precompiled packages

xoan, 2015-02-04:
Output for `$ ./bin/slicer --gui` in 1.2.5 and 1.2.6 in Debian Wheezy x86:

```
$ ./bin/slic3r --gui
Can't load '/home/user/Programas/Slic3r/lib/std/auto/OpenGL/OpenGL.so' for module OpenGL: /lib/i386-linux-gnu/libc.so.6: version `GLIBC_2.15' not found (required by /home/user/Programas/Slic3r/bin/../dll/libglut.so.3) at <content>/DynaLoader.pm line 157.
 at /home/user/Programas/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 119
Compilation failed in require at /home/user/Programas/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 119.
BEGIN failed--compilation aborted at /home/user/Programas/Slic3r/lib/std/Slic3r/GUI/Plater/2DToolpaths.pm line 119.
Compilation failed in require at /home/user/Programas/Slic3r/lib/std/Slic3r/GUI.pm line 16.
BEGIN failed--compilation aborted at /home/user/Programas/Slic3r/lib/std/Slic3r/GUI.pm line 16.
Compilation failed in require at (eval 107) line 1.
```

xoan, 2015-02-11:
I guess the lastest experimental precompiled packages are compiled against a newer libc6 version:

* https://packages.debian.org/wheezy/libc6

Should Debian Wheezy users wait until Jessie pass to "stable"?

lordofhyphens, 2015-02-11:
If you are using Debian, there are packages for 1.2.6 in debian
experimental. Use those or grab the source package and build your own.

alexrj, 2015-02-11:
This is the same as #861. Packages are built on Ubuntu 14.04 LTS. I don't know how to make them compatible with older glibc...

xoan, 2015-02-11:
Sorry for duplicating.

1.2.4-experimental works well on Debian Wheezy. Maybe libglut forces a newer symbol version in glibc, I don't know.

lordofhyphens, 2015-02-11:
For Wheezy, I recommend building the package from the pkg source in the
debian  experimental repo

xoan, 2015-06-17:
Fixed as in #2535

closed 2015-06-17 by xoan

alexrj, 2015-06-17:
Oh thank you @xoan for updating this.  So removing the GL/GLUT/whatever libraries fixed this one as well. Great!