slic3r/Slic3r #2628: Unify multiple 3d (and possibly 2d) views

Lenbok, 2015-02-05:
It's getting quite complicated from a UI perspective having two 2d views and two 3d views (the 3d views each have their own independent zoom and viewpoint). It would be nice if these were combined or simplified to some extent. (e.g.: along the lines of kisslicer, which gives more of the appearance of modes of a single view rather than multiple independent views).

Perhaps it would be as simple as synchronizing the viewpoint of 3d and 3d-preview tabs so that when switching between them, it looks more like changing modes rather than completely independent views.

lordofhyphens, 2015-02-05:
Keeping the current viewpoint between the 2 3d views should be sufficient
in my opinion.

I don't really use the 3d tool path preview much, though.

alexrj, 2015-02-15:
Good idea, I synchronized the viewport of the two 3D views. I agree with the other concerns about usability of the four views; the current state is temporary, it will definitely evolve towards something better also because of other planned changed in the plater view.

closed 2015-02-15 by alexrj

Lenbok, 2015-02-16:
That's quite a lot better. There's still a bit of a jump as you switch between the two 3d views (I guess due to the preview one having a slightly different sized view due to the layer slider), but I guess it's too hard to take that into account.