slic3r/Slic3r #2629: MAX Threads global setting

qharley, 2015-02-05:
Sent a profile to a friend, and it kept crashing his system.  Came down to my printing profile specifying too many threads.

We need a global maximum number of threads allowed even if the printing profile specifies more.  This could be in the <File><Preferences> menu.  If set to zero, it should take the specified number of threads as before.

whosawhatsis, 2015-02-05:
Is there any reason the number of threads shouldn't be a preferences setting rather than a profile setting?

lordofhyphens, 2015-02-05:
Unless there are side effects to the threadcount, I don't think so.

alexrj, 2015-02-05:
Yes @whosawhatsis, it should be there.  It's in the print settings because it's way older than the Preferences window itself :-)

qharley, 2015-02-05:
Sounds good!

harriv, 2015-02-05:
I think the real question here is why it crashed? There may be some problems in the code if specific thread count crashes the program.

qharley, 2015-02-05:
My guess is that it did not crash in slic3r... Used up all the ram, and
computer froze

alexrj, 2017-03-18:
Thread number was moved to application preferences. We can close this one. :)

closed 2017-03-18 by alexrj