slic3r/Slic3r #2631: No GUI 1.2.6 and dev (related to Wx)

xix-xeaon, 2015-02-05:
With 1.2.6 `./bin/slic3r --gui` gives me:
```
Can't load '/home/xixxeaon/Downloads/Slic3r-1.2.6/lib/std/auto/Wx/Wx.so' for module Wx: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory at <content>/DynaLoader.pm line 157.
 at /home/xixxeaon/Downloads/Slic3r-1.2.6/lib/std/Slic3r/GUI/AboutDialog.pm line 6
Compilation failed in require at /home/xixxeaon/Downloads/Slic3r-1.2.6/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /home/xixxeaon/Downloads/Slic3r-1.2.6/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /home/xixxeaon/Downloads/Slic3r-1.2.6/lib/std/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /home/xixxeaon/Downloads/Slic3r-1.2.6/lib/std/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 107) line 1.
```
I have the library at `/usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0` and I can run 1.2.5 just fine.

Not sure if it's related but I haven't been able to compile the gui in dev for a while because of Alien-wxWidgets-0.65. `sudo perl Build.PL --gui` gives me:
```
In file included from /usr/include/c++/4.9/type_traits:35:0,
                 from ../include/wx/strvararg.h:25,
                 from ../include/wx/string.h:46,
                 from ../include/wx/memory.h:15,
                 from ../include/wx/object.h:19,
                 from ../include/wx/wx.h:15,
                 from ../include/wx/wxprec.h:58:
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
In file included from ../include/wx/string.h:46:0,
                 from ../include/wx/memory.h:15,
                 from ../include/wx/object.h:19,
                 from ../include/wx/wx.h:15,
                 from ../include/wx/wxprec.h:58:
../include/wx/strvararg.h:345:18: error: ‘is_enum’ in namespace ‘std’ does not name a template type
     typedef std::is_enum<T> is_enum;
                  ^
../include/wx/strvararg.h:349:54: error: ‘is_enum’ was not declared in this scope
     enum { value = wxFormatStringSpecifierNonPodType<is_enum::value>::value };
                                                      ^
../include/wx/strvararg.h:349:68: error: template argument 1 is invalid
     enum { value = wxFormatStringSpecifierNonPodType<is_enum::value>::value };
                                                                    ^
Makefile:15504: recipe for target '.pch/wxprec_basedll/wx/wxprec.h.gch' failed
make: *** [.pch/wxprec_basedll/wx/wxprec.h.gch] Error 1
system: make all: 512 at ./Build line 60.
```

alexrj, 2015-02-15:
Is that OS X? What version?

xix-xeaon, 2015-02-16:
Ah, sorry, no it's Kubuntu 14.10 x64.

bogger33, 2015-03-01:
Having the same problem under LInux Mint Debian Edition libgtk2.0-0 is installed so it seems to be some sort of problem linking it properly? But I'm a noob so that's a pretty wild guess

curiouspl2, 2015-05-24:
hello, still broken in 1.2.7 : 
knoppix@Microknoppix:~/Desktop/slic3r/Slic3r.1.2.7/bin$ ./slic3r --gui
Can't load '/home/knoppix/Desktop/slic3r/Slic3r.1.2.7/lib/vrt/8b57d08e0d55cfe6498642b59ecf5ad1/Wx.so' for module Wx: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory at <content>/DynaLoader.pm line 157.
 at /home/knoppix/Desktop/slic3r/Slic3r.1.2.7/lib/std/Slic3r/GUI/AboutDialog.pm line 6
Compilation failed in require at /home/knoppix/Desktop/slic3r/Slic3r.1.2.7/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
BEGIN failed--compilation aborted at /home/knoppix/Desktop/slic3r/Slic3r.1.2.7/lib/std/Slic3r/GUI/AboutDialog.pm line 6.
Compilation failed in require at /home/knoppix/Desktop/slic3r/Slic3r.1.2.7/lib/std/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at /home/knoppix/Desktop/slic3r/Slic3r.1.2.7/lib/std/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 70) line 1.

i guess it's just libwx missing from the slicer binary package

alexrj, 2015-05-25:
The message says that libgtk-x11 is missing. I'm not sure the binary package should carry it. You can install it with your distribution's package manager.

alexrj, 2015-05-25:
@xix-xeaon, since new upstream packages were released you could try with `cpanm -v Alien::wxWidgets` and then with `cpanm -v Wx`. Let me know

xix-xeaon, 2015-05-26:
I'm now able to compile the wx packages, but I still get the same error with 1.2.6 and 1.2.7. The older version 1.2.5 still works, and the latest git gives me `Can't call method "config" on an undefined value at /home/xixxeaon/Downloads/Slic3r/lib/Slic3r/GUI/Tab.pm line 407.`, not sure what that is about.

alexrj, 2015-06-02:
@xix-xeaon, I have pushed a commit for that.  Can you please do a `git pull` and let me know if that fixes the issue? Thank you.

xix-xeaon, 2015-06-02:
Yeah, latest git now works =)

alexrj, 2015-06-02:
Good! Thank you!

closed 2015-06-02 by alexrj