slic3r/Slic3r #2632: Infill/Perimeter Overlap Overlapping Too Much (v. 1.2.6)

rf1k-mjh11, 2015-02-05:
The new "Infill/perimeters overlap" function seems to equate a value of 0% as a 50% overlap.

In other words, an overlap setting of 50% will position the infill bead at the exact same spot as the bead from the perimeter.  I would consider this a 100% overlap.

Playing around with the new function, I set the overlap to 65% and the preview functions of both Slic3r and Repetier Host showed an overlap of more than 100%, i.e. the infill was starting to show up outboard of the perimeter.
Conversely, an overlap value of -55% shows both the infill and perimeters still kissing each other.  Some may be the result of display, calculation or rounding errors.  But the two beads are barely touching, so I would consider that an overlap of 0%

I hope I explained this understandably. 

mjh11

Lord2Vader, 2015-02-10:
Yep can confirm that.
Just printing with 25% overlap and the perimeter is allmost completally printing over the infill!

alexrj, 2015-02-23:
Okay, I changed that.

closed 2015-02-23 by alexrj

epoxi, 2015-04-02:
Confirmed on *1.2.7-dev* also, but now differs from what @rf1k-mjh11 says:

**0% Overlap** - No overlap at all
[image]

**50% Overlap** - Almost there...
[image]

**100% Overlap** - Touching.
[image]
[image]

***115% Overlap*** - I think its the *15%* meant to be *default*
[image]
[image]