slic3r/Slic3r #2633: 1.2.6 x64 - Sequential printing not working

thiagopeixoto16, 2015-02-05:
It is always telling that the extruder will colide.
Ini and STL files on:
https://www.dropbox.com/sh/pqsmt90pyutrkib/AACOamcNYLzAAFOtbWWeA6zna?dl=0

This is the image:
[image]

zsolta, 2015-02-05:
I succeeded slicing them. No error message. Try moving the parts around a bit, that might help.

Slic3r 1.2.6 on Win7/64

thiagopeixoto16, 2015-02-06:
The Celtic piece is 60% scaled. Did you scaled too?

zsolta, 2015-02-06:
Nope, just put the parts on the plate, imported the config file and generated the g-code. I'll try the scaled version tomorrow.

zsolta, 2015-02-06:
It's already tomorrow and I just tried the scaled version. It works, no issues. I positioned the parts manually. Drag them a bit around, that helped me a few times in the past.

thiagopeixoto16, 2015-02-06:
The funny thing is that the problem is random...  
I tried again and no problems where found... I'm not sure how to reproduce now...
I'll share the configurations if I noticed this again.

alexrj, 2015-02-15:
I found the issue: the background processing were not restarted when a previous run failed because of object collision.

closed 2015-02-15 by alexrj