slic3r/Slic3r #2634: "Layer change G-code" layer_num placeholder out of order

forrestp, 2015-02-05:
I'm having some issues with the layer_num placeholder in the custom layer change g-code. When support is enabled, the layer_num outputs are duplicated and out of order. When support is disabled, everything works great. I can work around the duplication but not the incorrect ordering. I confirmed the issue exists on both windows and linux for versions 1.1.7 and 1.2.6

Details are below. Rather than post the entire gcode file, I used grep to filter the file so that only lines containing "before layer" remain. The ordering of the lines is preserved.

Config: Start with defaults, add "; before layer [layer_num]" to __Before layer change G-code__ and enable support material

STL: [calibration_cubev2.stl](http://www.thingiverse.com/download:631129)

Versions:
1. Windows 7 w/ Slic3r 1.1.7
2. Windows 7 w/ Slic3r 1.2.6
3. Ubuntu 14.04 w/ Slic3r 1.1.7
4. Ubuntu 14.04 w/ Slic3r 1.2.6

Filtered output gcode without support (Win7, Slic3r 1.2.6 GUI):
; before layer 0
; before layer 1
; before layer 2
; before layer 3
; before layer 4
; before layer 5
; before layer 6
; before layer 7
; before layer 8
; before layer 9
; before layer 10
; before layer 11
; before layer 12
; before layer 13
; before layer 14
; before layer 15
; before layer 16
; before layer 17
; before layer 18
; before layer 19
; before layer 20
; before layer 21
; before layer 22
; before layer 23
; before layer 24
; before layer 25
; before layer 26
; before layer 27
; before layer 28
; before layer 29
; before layer 30
; before layer 31
; before layer 32
; before layer 33
; before layer 34
; before layer 35
; before layer 36
; before layer 37
; before layer 38
; before layer 39
; before layer 40
; before layer 41
; before layer 42
; before layer 43
; before layer 44
; before layer 45
; before layer 46
; before layer 47
; before layer 48
; before layer 49

Filtered output gcode with support (Win7, Slic3r 1.2.6):
; before layer 0
; before layer 0
; before layer 1
; before layer 1
; before layer 2
; before layer 2
; before layer 3
; before layer 3
; before layer 4
; before layer 4
; before layer 5
; before layer 5
; before layer 6
; before layer 7
; before layer 6
; before layer 8
; before layer 7
; before layer 9
; before layer 8
; before layer 9
; before layer 10
; before layer 10
; before layer 11
; before layer 12
; before layer 11
; before layer 13
; before layer 12
; before layer 14
; before layer 15
; before layer 13
; before layer 14
; before layer 16
; before layer 15
; before layer 17
; before layer 16
; before layer 18
; before layer 17
; before layer 19
; before layer 18
; before layer 19
; before layer 20
; before layer 20
; before layer 21
; before layer 21
; before layer 22
; before layer 22
; before layer 23
; before layer 23
; before layer 24
; before layer 24
; before layer 25
; before layer 25
; before layer 26
; before layer 26
; before layer 27
; before layer 27
; before layer 28
; before layer 29
; before layer 28
; before layer 30
; before layer 29
; before layer 31
; before layer 30
; before layer 32
; before layer 31
; before layer 33
; before layer 34
; before layer 32
; before layer 35
; before layer 33
; before layer 36
; before layer 34
; before layer 37
; before layer 35
; before layer 36
; before layer 38
; before layer 37
; before layer 39
; before layer 40
; before layer 41
; before layer 42
; before layer 43
; before layer 44
; before layer 45
; before layer 46
; before layer 47
; before layer 48
; before layer 49

forrestp, 2015-02-09:
Just realized this is probably due to Slic3r tracking support material layers separately from regular layers. 

Would it be possible to have a 'layer_type' placeholder available in the custom layer change gcode fields as well? It could return 'support' or 'object' depending on what type of layer it is. That would solve my problems.

alexrj, 2015-02-15:
Yeah, that's because support layers have their own number. Maybe the right thing is just to have a single progressive number as the user would expect...

alexrj, 2015-05-03:
Fixed! Thank you.

closed 2015-05-03 by alexrj

forrestp, 2015-05-04:
Thanks for fixing this!