slic3r/Slic3r #2635: Support material regression

markfinn, 2015-02-06:
I have an object with stand-offs that hang down unsupported like stalactites which are no longer supported correctly since 379cde30e.

Prior to that commit, the first interface layer of support was, well supported:
[image]
[image]

And after that commit, it just floats in the air:
[image]
[image]

There is another issue where the top of the screw hole in this same standoff tries to get support, which isn't needed since it can bridge.  Again, this changed (for the worse) at the same commit.

markfinn, 2015-02-06:
I have the stl available [here](https://github.com/markfinn/hackerspace_access_control/blob/teststl/newcase.fcstd?raw=true)

config:
# generated by Slic3r 1.2.2-dev on Thu Feb  5 16:03:29 2015
avoid_crossing_perimeters = 0
bed_shape = -109x-142.5,109x-142.5,109x142.5,-109x142.5
bed_temperature = 100
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = \nG92 E0 ; e to 0\nG1 F600.000 E-2 ; retract more\nM190 S0     ; bed heat off\nM104 S0 ; turn off temperature\nM107 ; fan off\nG91; Use Relative Coordinates\nG1 Z3 ; raise Z a little\nG90; Use Absolute Coordinates\nG1 X-200 Y400 ; move bed out\nM84     ; disable motors
external_perimeter_extrusion_width = 0
external_perimeter_speed = 100%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 2.88
fill_angle = 45
fill_density = 20%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 110
first_layer_extrusion_width = 200%
first_layer_height = 100%
first_layer_speed = 30%
first_layer_temperature = 240
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_speed = 80
interface_shells = 0
layer_gcode = 
layer_height = 0.32
max_fan_speed = 100
min_fan_speed = 30
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.4
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 50
perimeters = 3
post_process = 
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 1
retract_length = 0.7
retract_length_toolchange = 3
retract_lift = 0
retract_restart_extra = 0.05
retract_restart_extra_toolchange = 0
retract_speed = 30
seam_position = random
skirt_distance = 6
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 15
small_perimeter_speed = 30
solid_fill_pattern = rectilinear
solid_infill_below_area = 70
solid_infill_every_layers = 7
solid_infill_extrusion_width = 0
solid_infill_speed = 80
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 Y; home\nG28 Z; home\nG1 Z5 ; raise Z a little
support_material = 1
support_material_angle = 0
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 10
support_material_speed = 60
support_material_threshold = 45
temperature = 229
thin_walls = 1
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 60
top_solid_layers = 3
travel_speed = 130
use_firmware_retraction = 1
use_relative_e_distances = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0

markfinn, 2015-02-06:
From what I've seen, this may be the same issue as the first picture in issue #2618

lordofhyphens, 2016-06-28:
@markfinn Still a problem in 1.3.0-dev?

lordofhyphens, 2016-12-30:
Your issue is related to your overhang angle. Letting Slic3r figure it out supports those areas.
With overhang threshold = 0:
[image]

With overhang threshold = 45
[image]

Config/stl: [2635.zip](https://github.com/alexrj/Slic3r/files/678988/2635.zip)

lordofhyphens, 2017-03-28:
With the new syntax for automatic support generation, can use 150% or such to reduce the effective angle.