slic3r/Slic3r #2636: Possible bridging or bridge detection regression

Vicious-one, 2015-02-06:
Strange bridge region detection was encountered when trying to bridge the fins:
[image]

1.2.0 binary:
[image]

Current:
[image]

Vicious-one, 2015-02-16:
When trying to avoid the problem by shifting bridges to different layers, I've also noticed than bridge infill pattern angle is not applied to the bridges besides the leftmost one:
[image]
[image]

Vicious-one, 2015-02-19:
[image]

Vicious-one, 2015-02-26:
Seems that bridging infill angle is applied only to one of the bridges:
[image]
[image]
[image]
[image]

Vicious-one, 2015-02-26:
Processing bridge at layer 46:
Angle -1.000000:
Processing bridge at layer 46:
Angle 1.570795:
Processing bridge at layer 106:
Angle 1.396263:
Processing bridge at layer 106:
Angle -1.000000:

BridgeDetector::detect_angle() sets angle to -1 which results in applying default infill angle for current layer.

Vicious-one, 2015-02-27:
@alexrj -- please revert 379cde30e296482e784825e9df586fe654023c4 so everyone can have proper multiple bridges again!

alexrj, 2015-02-27:
I fixed that regression. Thank you for reporting it :-)

closed 2015-02-27 by alexrj

Vicious-one, 2015-02-27:
Thanks, now it seems to be working as before.
One question remains, though: the calculated angle seems strange, doesn't it?
[image]
[image]

Vicious-one, 2015-02-28:
Okay, having seen the best angle calculation algorithm, the does not seem strange anymore, but the fact that they are not symmetric still does.

Never mind though, I'll just play a little with detection method.

Vicious-one, 2015-02-28:
I've changed line interval to 1/4 extrusion width and consistently getting more sane results.

(double line_increment = this->extrusion_width/4.0; BridgeDetector.cpp:116)
[image]
[image]
[image]
[image]
[image]