slic3r/Slic3r #2637: I think that somewthing is wrong in 1.2.6

migui88, 2015-02-06:
Hi!
Yesterday I printed this file and I found something strange. I discovered that there are missing layer. The right part is printed using the last stable slic3r 1.1.7 and the left part is using 1.2.6.  
[image]

I put the STL, the Gcode generated and the Slic3r config in MEGA

https://mega.co.nz/#!FN4BQBoS!8lovWgifAayCNdTGlYV5_krf_4bSB5Ecqcfda2H0IBk

Louis-Cypher, 2015-02-07:
I strongly support this issue. Sadly I do not have the part anymore (maybe I can find it in the trash on monday) but I can provide the gcode if meaningful. I printed a very simple flange and many "tracks" of material have been missing. I stopped the print like some previous others. I used the same parameters I have used successfully in 1.2.5 for the same part.

Lenbok, 2015-02-07:
@migui88 The STL and gcode files you attached do not appear to be the same ones as in the photo.

migui88, 2015-02-08:
@Lenbok Sorry! I put other part in the zip. Here is the right gcode and the stl. 

https://mega.co.nz/#!FdAlUaIQ!zAHOo3OpqKYJTa_yol10RESzp0IGJZM0Q036CtKD7-4

alexrj, 2015-02-09:
Actually, this is how your G-code looks like in Repetier-Host's preview:

[image]

It looks correct. Last layer is at Z = 58.3mm, which is the height of the STL file.
Check your printer, you have probably mechanical issues or Z driver overheating, or too high max_speed or max_acceleration limits in your firmware.

closed 2015-02-14 by alexrj