slic3r/Slic3r #2639: Slic3r 1.2.6 gives shortcutting perimeters

hyperair, 2015-02-07:
I'm not sure what causes this, but I just discovered some odd perimeters generated from a drill bit sharpening jig I was making.

[image]
Gcode: https://gist.github.com/554bdfb70695d89c66fc
Config: https://gist.github.com/6f97fcb7641a43fdf79b
STL: https://dl.dropboxusercontent.com/u/169656/sharpening-jig.stl

hyperair, 2015-02-07:
I verified the generated G-code in gcode.ws, so it's not an issue with the preview.

justmyopinion, 2015-02-07:
I do not know what is causing the trouble, but if you rotate item  (45deg), problem seems to disappear.

Lenbok, 2015-02-08:
I see it too. I tried slicing with 1.2.5 and the perimeters are fine, so something between 1.2.5 and 1.2.6.

hyperair, 2015-02-08:
Hmm, interesting workaround. Thanks!

lordofhyphens, 2015-02-08:
Any idea if this problem is in 1.2 5 or 1.2.4?

hyperair, 2015-02-08:
Don't see it in 1.2.5.

alexrj, 2015-02-15:
Thank you. That looks like an issue in the Clipper library. I isolated the test case and escalated the report to the kind Clipper author:
https://sourceforge.net/p/polyclipping/bugs/126/

AngusJohnson, 2015-02-16:
Yes, there was a bug in the Clipper code that has now been fixed in [the latest revision on SF](https://sourceforge.net/p/polyclipping/code/HEAD/tree/trunk/).

alexrj, 2015-02-16:
Thank you @AngusJohnson for coming here :) I'm glad to contribute to achieve more robustness in the already wonderful Clipper library.

commit edbc1147 (2015-02-22): Update to Clipper 6.2.9. Fixes #2639
closed 2015-02-22 by alexrj