slic3r/Slic3r #2640: [feature suggestion] 2D Layer Modifiers

Vicious-one, 2015-02-08:
There's one feature I've not yet seen in any slicing software:

*Manually-defined (drawn bu a mouse with lines or bezier curves) 2D per-layer polygon modifiers.*

They can work exactly like 3D Modifier meshes, but with a single slice only.

If a user encounters a show-stopping bug or a suboptimal(segmented) infill, he just draws a polygonal shape with a mouse, which overrides perimeter and infill settings, disables bridge detection, etc. directly in the toolpath window.

dexy31337, 2015-02-09:
I'm almost sure this feature isn't going to be implemented, as it requires too much code to be written, which shouldn't be in slicing software. Also, "slice layer" comes after slicing is done, so it's hard to specify what exactly you want to be generated, especially if you'll want to modify something like layer height.
Also, I believe that this task is done perfectly with modifier meshes, in 3d editing software, where it should belong IMHO. Just make a mesh with a height of 1 layer, depending on your printing preferences and place it on a corresponding height.

Vicious-one, 2015-02-09:
@dexy31337 Yes, this is more of a post-processing than slicing. Currently I have to resort to serious g-code hacking when slicing goes wrong *again*. In fact, it does go wrong nearly every time when bridging tricky gaps or in narrow areas. 

Yes, one might say I should taylor the models for my slicing software to avoid its army of bugs, but seriously...

Anyway, I think this kind of tool, well-integrated into slicing software (using its profiles, etc), is going to be invaluable considering ease of use and efficiency. Just sayin...

lordofhyphens, 2015-02-09:
Well, we run on something of a do-ocrocy (with Sound having veto powers)
here. If you want this feature and nobody else wants to work on it, you
will probably have to do it yourself and submit a pull request.

victoriansir, 2015-02-13:
I support this feature, in some situations it would be very usefull

lordofhyphens, 2017-03-19:
We have modifier meshes now that are movable and controllable, so set "slab" to your layer height and you have your "2d" to cover that part.