slic3r/Slic3r #2646: Peculiar & Inconsistent Bridging

ednisley, 2015-02-09:
Recent versions (through 0d822f0) produce unusual (and unusable) bridges on my Chain Mail models. This seems related to other bridging problems, but in these cases, identical shapes produce different results; perhaps that can help track down what's going wrong.

As nearly as I can tell, the bottom layer of all bridges _generally_ uses the infill pattern for the current layer, rather than the direct-across bridge pattern. This doesn't come from a difference in the models, because all of the chain mail links come from the same OpenSCAD object and are identical.

For example, three of these four bridges use infill across the link bars:

[image]

The single correct bridging pattern spans the far right-rear bar in a larger group of links:

[image]

The first layer of bridging for the "armor" caps on the 2x2 array of links in the middle shows both infill and bridging patterns:

[image]

The bridging in the right rear cap combines _both_ patterns, with a few bridge threads spanning the left edge and diagonal infill across the remainder.

The same type of problem appears with the default Slic3r configuration, as well as my customized MakerGear M2 configuration. Of course, the exact results vary unpredictably with the exact values for all the perimeter / infill parameters. I have yet to find a combination that produces the same results as the older versions that let me produce large chain mail sheets with uniformly good bridges:

[image]

That sheet was printed in late November 2014. The comment at the top of the G-Code file reads:

    ; generated by Slic3r 1.2.1 on 2014-11-27 at 16:41:09

Slic3r doesn't report any problems with these STL files; they're all manifold. The models include slight (on the order of 0.05 mm) protrusions or recesses at the openings to eliminate coincident edges and surfaces, but changing / eliminating those doesn't affect the  bridging problems.

The parameters at the top of the OpenSCAD program provide fine-grained control over the exact model dimensions and chain mail link attributes.

This is done with Xubuntu 14.04LTS and compiled-from-source versions of OpenSCAD through 37421bb.

The OpenSCAD, STL, and Slic3r config.ini are in:
https://www.dropbox.com/s/nm20qqbjipum10e/Slicer%200d822f0%20-%20bridging.zip?dl=0

If I can run any other tests to help track down the problem, let me know!

Thanks ...

Ghoughpteighbteau, 2015-02-19:
I'm having difficulties with this as well. Slicer makes odd choices when it's bridging over unsupported area's. It should span gaps as much as possible.

In this case in particular I'd love it if slicer just used a rectilinear pattern, because it would work!

[image]

STL's to test on: 

https://www.dropbox.com/s/darjs2byyex5zis/testbricklego2.stl?dl=0
https://www.dropbox.com/s/u06sb01oz1r89k3/testbricklego.stl?dl=0

MadMarkTheSane, 2015-02-21:
# Simpler Test Case and Possible Work Around
Firstly I think I have identified which version introduced this problem.  Here is a simpler test model which also exhibits the same fault sliced with version 1.2.1:
[image]
And the same model sliced with version 1.2.2:
[image]
As Ed orignialy reported it seems that only one bridge is being detected per layer.
I can confirm this bug exists in versions 1.2.2 to 1.2.6, windows x64 and linux x64 builds from the web site.  All tests were done with a 'default' configuration by clearing out slic3r's config directory between each test.
## Possible workaround
If the model contains multiple solids and you can 'split' the the model using the toolbar then versions 1.2.2 to 1.2.6 correctly generate the bridge directions again.
[image]
Of course this wont help if the model is a single solid.  I'm not sure if it will help in your case Ed, as I think once split your chain mail parts will be re-arranged and become unlinked.
I would seem that slic3r is no longer detecting multiple bridges on the same layer and optimising directions for them which gave me a thought:
[image]
**It breaks the frame vertex test case as shown on the website!**
## Link to my test files and config
http://1drv.ms/1Av7qxc

ednisley, 2015-02-21:
> once split your chain mail parts will be re-arranged

All the links remain in place after splitting the mesh! Clicking Arrange does exactly what you'd expect, but until then nothing moves.

Unfortunately, every link has two bars that require bridges on the same level and, even though they're parallel, one bar still gets the infill thread orientation.

The larger armor caps, however, all bridge perfectly, so the problem happens only with multiple bridges on the same layer within a single object.

Most of my designs look like brackets: a single object with multiple bridges. If I can gimmick each bridge onto a separate layer, then your workaround will dodge the problem until an official fix arrives.

Thanks!

MadMarkTheSane, 2015-02-21:
Happy to have helped a bit at least!

Seems like we should have enough information on this issue here for someone who knows what they are doing to take a look at it.  If I knew the first thing about Perl I might be tempted to take a look at it myself but unfortunately when it gets above the C/C++ level I'm way out of my comfort zone!

Ghoughpteighbteau, 2015-02-23:
At least in my case, the ability to disable the bridging algorithm would be a nice stopgap.

alexrj, 2015-02-27:
This regression was fixed in #2636. Thank you for reporting it.

closed 2015-02-27 by alexrj