slic3r/Slic3r #2647: Slic3r crashes linux

prof-braino, 2015-02-09:
Don't know if this is the right place to report this, thanks for you patience: Installed slic3r about a week or two ago, so it should be current.  Unfortunately yesterday it began crashing the PC when launched; PC freezes, mouse and unscreen clock stop, PC reboots.  
Dell Pentium 4, Linux Mint 16-32 MATE, Nvidia graphic card nouveau driver
During install I recieved an error that NUMPY was missing, I installed python-numpy
The build-GUI option did not work, something with Wx.
I saw a report about an error in the Nvidia Nouveau driver that I cannot find at the moment; I did not want to go down the graphics card driver rat-hole if it could be avoided.
I have been slic3r-ing for a couple weeks, this crashing the PC started a couple days ago.  
I am a  user not a developer, all I have to changing is print speeds and filament size etc.

lordofhyphens, 2015-02-09:
I recommend using the debian or Ubuntu packages for slic3r rather than the
build scripts. Get the version in debian experimental if you want 1
2.6.

I  can provide a copy of the source package and some limited instructions
if you want to take a go at building your own.

prof-braino, 2015-02-09:
OH, I see. The instructions are confusing, I thought the install from git WAS the way to get the precompiled pakages (but none existed for linux and we need to compile from git).   I will open a separate issue to request better instruction for installing the precompiled package on  linux. 
However, I'll leave this open since the issue still exists that the version from git crashes linux mint 32 bit.

lordofhyphens, 2015-02-09:
Okay.

The build script works thusly:

First you need the build dependencies.

The build script grabs a bunch of stuff with cpanm, which is black perl
magic. Then it compiles libslic3r and installs that somewhere.

The --gui tries to grab more packages via cpanm.

Almost everything that cpanm grabs is available in system lib packages
anyway.

If you are getting crashes;  we need to see a copy of your configuration
and stls that you have seen the problem with.

If you have gdb installed and the crash produced a core file, very helpful
to run gdb /usr/bin/perl core and then run the back command. Paste the
output here, is helpful to trace what is going on.

Additionally, you are running most recent commit, so 1.2.6-dev, when you
get from commit.

If you used git clone to download the source,  you can run git checkout
1.2.6 to get the release source for 1.2.6 and build again.

lordofhyphens, 2015-02-09:
Please edit the title of this issue to add 1.2.6-dev so we know which
version to look for.

prof-braino, 2015-02-10:
Changed title to 1.2.6-dev, FYI the date stamp on all the files is Feb 2 2015.

harriv, 2015-02-10:
If PC reboots, the problem is most probably on the OS or hardware level.

prof-braino, 2015-02-13:
Changing to the pre-compiled version 
http://dl.slic3r.org/linux/slic3r-linux-x86-1-1-7-stable.tar.gz
fixed the issue.
Clarifying the instructions to installing the precompiled verion would address the root cause.

lordofhyphens, 2015-02-13:
That's not the precompiled 1.2.6, that's the precompiled 1.1.7 (which is
much older and has different bugs).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2015-02-13:
The website at http://slic3r.org/ has a very clear "Download" button which points to the binary pages. The project home page at GitHub says clearly: *"You can just download a precompiled package from slic3r.org; it will run without the need for any dependency."* So it's quite clear…

Also, I'm a bit confused by the report. Slic3r is not written in Python and it doesn't use numpy.  If you want to try the precompiled 1.2.6 package and find problems with it, any issue report is welcome.

closed 2015-02-13 by alexrj