slic3r/Slic3r #2648: *.stl file "no layers detected" in Slic3r

finlock, 2015-02-09:
I drag and drop *.stl files into Slic3r.  These *.stl files are exported from my preferred cad system (geomagic design).  I get the following error when I try to generate gcode in Slic3r: "No layers were detected.  You might want to repair your *.stl file(s) or check their size and retry".  Also:  The image displayed by Slic3r is just a small circle and does not represent my actual drawing.  I have loaded these same files into Netfabb and no errors show (and Netfabb shows the proper dwg).  I have tried both (1) just exporting out of Netfabb and (2) repairing, applying repair and exporting and have gotten the same results.  Note:  I only have this problem with *.stl files from geomagic design and not other programs (that said, I prefer to use geo for other reasons).  Any help is greatly appreciated!

dexy31337, 2015-02-09:
@finlock please provide problematic stl files to reproduce problem

finlock, 2015-02-11:
Ok, thank you for reviewing...here's the ascii stl:

solid ascii
  facet normal 9.951847e-001 -9.801714e-002 2.348212e-017
    outer loop
      vertex   -1.534929e-001 3.053164e-002 0.000000e+000
      vertex   -1.565000e-001 1.916572e-017 1.600000e-001
      vertex   -1.565000e-001 -1.916572e-017 0.000000e+000
    endloop
  endfacet
  facet normal 9.951847e-001 -9.801714e-002 0.000000e+000
    outer loop
      vertex   -1.534929e-001 3.053164e-002 0.000000e+000
      vertex   -1.534929e-001 3.053164e-002 1.600000e-001
      vertex   -1.565000e-001 1.916572e-017 1.600000e-001
    endloop
  endfacet
  facet normal 9.569403e-001 -2.902847e-001 0.000000e+000
    outer loop
      vertex   -1.534929e-001 3.053164e-002 0.000000e+000
      vertex   -1.445871e-001 5.988996e-002 0.000000e+000
      vertex   -1.534929e-001 3.053164e-002 1.600000e-001
    endloop
  endfacet
  facet normal 9.569403e-001 -2.902847e-001 0.000000e+000
    outer loop
      vertex   -1.534929e-001 3.053164e-002 1.600000e-001
      vertex   -1.445871e-001 5.988996e-002 0.000000e+000
      vertex   -1.445871e-001 5.988996e-002 1.600000e-001
    endloop
  endfacet
  facet normal 8.819213e-001 -4.713967e-001 0.000000e+000
    outer loop
      vertex   -1.445871e-001 5.988996e-002 1.600000e-001
      vertex   -1.445871e-001 5.988996e-002 0.000000e+000
      vertex   -1.301250e-001 8.694674e-002 0.000000e+000
    endloop
  endfacet
  facet normal 7.730105e-001 -6.343933e-001 0.000000e+000
    outer loop
      vertex   -1.301250e-001 8.694674e-002 1.600000e-001
      vertex   -1.301250e-001 8.694674e-002 0.000000e+000
      vertex   -1.106622e-001 1.106622e-001 0.000000e+000
    endloop
  endfacet
  facet normal 6.343933e-001 -7.730105e-001 0.000000e+000
    outer loop
      vertex   -1.106622e-001 1.106622e-001 1.600000e-001
      vertex   -1.106622e-001 1.106622e-001 0.000000e+000
      vertex   -8.694674e-002 1.301250e-001 0.000000e+000
    endloop
  endfacet
  facet normal 4.713967e-001 -8.819213e-001 0.000000e+000
    outer loop
      vertex   -8.694674e-002 1.301250e-001 1.600000e-001
      vertex   -8.694674e-002 1.301250e-001 0.000000e+000
      vertex   -5.988996e-002 1.445871e-001 0.000000e+000
    endloop
  endfacet
  facet normal 2.902847e-001 -9.569403e-001 0.000000e+000
    outer loop
      vertex   -5.988996e-002 1.445871e-001 1.600000e-001
      vertex   -5.988996e-002 1.445871e-001 0.000000e+000
      vertex   -3.053164e-002 1.534929e-001 0.000000e+000
    endloop
  endfacet
  facet normal 9.801714e-002 -9.951847e-001 0.000000e+000
    outer loop
      vertex   -3.053164e-002 1.534929e-001 1.600000e-001
      vertex   -3.053164e-002 1.534929e-001 0.000000e+000
[... 3493 lines cut ...]
      vertex   -2.355893e-001 9.758428e-002 -6.245699e-017
    endloop
  endfacet
  facet normal -6.097137e-016 1.849546e-016 -1.000000e+000
    outer loop
      vertex   2.355893e-001 -9.758428e-002 -6.245699e-017
      vertex   1.445871e-001 -5.988996e-002 0.000000e+000
      vertex   2.501002e-001 -4.974803e-002 -6.245699e-017
    endloop
  endfacet
  facet normal -1.849546e-016 -6.097137e-016 -1.000000e+000
    outer loop
      vertex   9.758428e-002 2.355893e-001 -6.245699e-017
      vertex   3.053164e-002 1.534929e-001 0.000000e+000
      vertex   4.974803e-002 2.501002e-001 -6.245699e-017
    endloop
  endfacet
  facet normal 1.849546e-016 -6.097137e-016 -1.000000e+000
    outer loop
      vertex   -5.988996e-002 1.445871e-001 0.000000e+000
      vertex   -9.758428e-002 2.355893e-001 -6.245699e-017
      vertex   -4.974803e-002 2.501002e-001 -6.245699e-017
    endloop
  endfacet
endsolid

lordofhyphens, 2015-02-11:
It's better to upload the STL to a file locker website and post a link.
On Feb 11, 2015 5:43 AM, "finlock" <notifications@github.com> wrote:

> Ok, thank you for reviewing...here's the ascii stl:
>
> solid ascii
> facet normal 9.951847e-001 -9.801714e-002 2.348212e-017
> outer loop
> vertex -1.534929e-001 3.053164e-002 0.000000e+000
> vertex [...]

finlock, 2015-02-11:
Sorry, I'm newer to this.  Are you able to point me towards a couple file locker sites?

Thanks, Mike

Date: Wed, 11 Feb 2015 05:35:54 -0800
From: notifications@github.com
To: Slic3r@noreply.github.com
CC: varchive@hotmail.com
Subject: Re: [Slic3r] *.stl file "no layers detected" in Slic3r (#2648)

It's better to upload the STL to a file locker website and post a link.

On Feb 11, 2015 5:43 AM, "finlock" <notifications@github.com> wrote:

> Ok, thank you for reviewing...here's the ascii stl:

>

> solid ascii

> facet normal 9.951847e-001 -9.801714e-002 2.348212e-017

> outer loop

> vertex -1.534929e-001 3.053164e-002 0.000000e+000

> vertex -1.565000e-001 1.916572e-017 1.600000e-001

> vertex -1.565000e-001 -1.916572e-017 0.000000e+000

> endloop

> endfacet

> facet normal 9.951847e-001 -9.801714e-002 0.000000e+000

> outer loop

> vertex -1.534929e-001 3.053164e-002 0.000000e+000

> vertex -1.534929e-001 3.053164e-002 1.600000e-001

> vertex -1.565000e-001 1.916572e-017 1.600000e-001

> endloop

> endfacet

> facet normal 9.569403e-001 -2.902847e-001 0.000000e+000

> outer loop

> vertex -1.534929e-001 3.053164e-002 0.000000e+000

> vertex -1.445871e-001 5.988996e-002 0.000000e+000

> vertex -1.534929e-001 3.053164e-002 1.600000e-001

> endloop

> endfacet

> facet normal 9.569403e-001 -2.902847e-001 0.000000e+000

> outer loop

> vertex -1.534929e-001 3.053164e-002 1.600000e-001

> vertex -1.445871e-001 5.988996e-002 0.000000e+000

[... 3486 lines cut ...]
> vertex 7.270985e-002 -1.755371e-001 1.310000e-001

> endloop

> endfacet

> facet normal 6.343933e-001 -7.730105e-001 -2.830981e-016

> outer loop

> vertex 1.055583e-001 -1.579792e-001 2.900000e-002

> vertex 1.343503e-001 -1.343503e-001 2.900000e-002

> vertex 1.055583e-001 -1.579792e-001 1.310000e-001

> endloop

> endfacet

> facet normal 7.730105e-001 -6.343933e-001 -1.415491e-016

> outer loop

> vertex 1.343503e-001 -

dexy31337, 2015-02-11:
@finlock maybe [Dropbox](https://db.tt/ugI1LKL) ?

alexrj, 2015-02-14:
Your model is 0.16mm thick.
Since you haven't read the guidelines and haven't posted enough information I can just guess that you're using a thicker layer height, so no layers can be fitted in a 0.16mm object (of course).

closed 2015-02-14 by alexrj