slic3r/Slic3r #2649: 1.2.6: Slic3r preview doesn't render skirt and brim

siberx, 2015-02-09:
I'm absolutely loving the new toolpath preview feature - it's a big timesaver when sanity checking generated prints before starting them.

I did notice, however, that brim and skirt do not appear to render in the previewed toolpath:
Preview in question:
[image]
First layer of the above print:
[image]

I'm using the following Skirt/brim settings:
Loops (minimum): 1
Distance from object: 5mm
Skirt height: 1 layers
Minimum extrusion length: 0mm

Brim width: 2mm

I tested with a taller skirt height thinking it might just be the first layer not showing correctly, but I still see no skirt even with a height of 10 layers. I am making an *assumption* that the toolpath preview should include these elements, but that makes sense to me considering it's a "toolpath" preview and should thus show the full path of the tool ;)

a4jp-com, 2015-02-09:
If possible, I'd like to see both the brim and skirt as well.

jonaskuehling, 2015-02-16:
My support for this request!

3D preview should render all paths that will effectively be printed, including skirt, brim etc.

alexrj, 2015-02-16:
This is definitely in the roadmap... I just have one concern about tall skirts that would hide the object entirely...

jonaskuehling, 2015-02-16:
Yeah, thought about that the second I hit "comment"... Maybe brim/skirt could be rendered with transparency..

gege2b, 2015-02-16:
Or maybe rendered with verry thin lines on layers >= 2 ?

jonaskuehling, 2015-02-16:
Mh, I think every to-be-printed toolpath should be visualised, since otherwise no proper diagnosis is possible if the configured settings have been applied like intended.

gege2b, 2015-02-16:
Well, I agree with you for the object, the brim and the support, as all of this is directly connected to the object

My concern was more about visualizing the entire skirt. As @alexrj said, if the skirt is tall, it would hide the object behind

As long as the skirt is never connected the object itself, I think showing only his X first layer(s) should be enough

Render the skirt with transparency could be a solution, but I'm affraid it would make the 3D Preview/Toolpath slower (right now it's not always really reactive)

gege2b, 2015-02-16:
Another option could be to add a switch button or a checkbox around the 3D preview/toolpath to show / hide skirt on the fly

jonaskuehling, 2015-02-16:
Yes, a toggle button would be the most favoured solution for me. Could even be inside the preview window as overlay buttons, think of Cura for example..

(Could think of a "parallel projection" toggle switch in the 3D plater tab as well, and maybe shortcut buttons for top/front/side/bottom view angle - see Repetier Host)

a4jp-com, 2015-02-16:
If you are worries about the skirt or brim hiding the model just add a checkbox or sliding switch to hide them. Simple.

alexrj, 2015-05-17:
Done.

closed 2015-05-17 by alexrj