slic3r/Slic3r #2651: Please don't use "Build.PL" this way.

kentfredric, 2015-02-11:
Build.PL is a special name within the CPAN/Perl ecosystems and all such Build.PL's are expected to conform to the Module::Build API, and no automatic "install with cpanm" stuff is permitted.

( Similarly, Makefile.PL is expected to conform to the EUMM API and emit a makefile with various useable targets )

This code thus also violates user choice by running tests regardless, whereas with an official Module::Build based toolchain, the user can run tests or not at their choice.

And the violation of seperation between "configure", "build", "install" and "test" phases makes it needlessly difficult for source-based-distribution downstreams to ship this.

For instance, somebody at Gentoo wants to package this. Your custom Build.PL at present makes this needlessly difficult. Gentoo users themselves can elect to run tests.... and that test run happens in a dedicated phase in the installation cycle.

Present Build.PL breaks this.

swiftgeek, 2015-02-11:
You can look at this abomination of a package :D (only prepare, build, check and package functions)
https://aur.archlinux.org/packages/sl/slic3r-git/PKGBUILD

alexrj, 2015-02-11:
I'm a CPAN author since 2003, and I maintain a number of modules.
In this case you're missing one point: Slic3r **is NOT** a Perl module distribution.  It's not in CPAN.  It's a standalone application that is not required to obey any API since it's not meant to be installed by a CPAN client.

If a distribution maintainer wants to use a CPAN build script, it's trivial to write one.

There are a number of good reasons about why Slic3r supplies a script that uses `cpanm` (and works very well). The main one is that CPAN is sadly broken on most distributions out there, and back in 2011 I was spending a lot of time troubleshooting users' problems with their broken distributions, and explaining that Perl doesn't suck. Since my time is limited, I want to spend time on coding rather on troubleshooting and I chose a solution that allegedly works for everybody.

kentfredric, 2015-02-11:
I Understand the intent of the script =). And I understand how for novice users *what* it does is useful. However, my complaint is more along the lines of what the file is called, re: calling it Build.PL instead of something else.

Calling it Build.PL gives false assumptions to consumers and maintainers alike who see the source tree and go "Sweet, Build.PL, ./Build .... " then getting explosions.

You could say the presence of that specific Build.PL triggers a lot of heuristic based logic, that is correct for 99.999% of the cases where that file is seen, but this case false-positives the "Valid MB-Style dist" heuristic.

And even here, downstream aren't meaning to be a "CPAN Client", but those heuristics are invariably baked in which have to be circumvented and special cased for this component.

I would even potentially suggest shipping a top-level `Build.PL` in the dist itself somewhere in a dir like "maintainers/" which **could** work in a CPAN-style toolkits could simply perform an early step like 

   `cp maintainers/Build.PL Build.PL` 

And then apply standard CPAN style conventions. ( And this would greatly make downstreams life easier by having an upstream-supported install system that is compatible with their own, instead of all downstreams having to reinvent your tooling )

swiftgeek, 2015-02-11:
I just do bunch of exports like in any other cpan package and it works fine, separate prove makes it nicer for displaying and easier to embed.

The only additional thing specific to slic3r is """ export SLIC3R_NO_AUTO="true" """
Maybe it should be noted in readme that it exists

alexrj, 2017-03-18:
I'm closing this one. Slic3r is becoming less and less a Perl project, and less suitable for installation in a CPAN environment. So far, nobody reported confusion about wrong assumptions for the Build.PL syntax.
We will switch to CMake soon anyway :-)

closed 2017-03-18 by alexrj