slic3r/Slic3r #2652: [Feature Request] Cut with cubic alignment post

troffgopher, 2015-02-11:
For a given part it's sometimes more reliable to cut a part to produce multiple pieces with flat sides that require no support and then glue the printed parts back together. Aligning those parts can be tricky.

With the cut tool, why not allow a void to be created in each piece with a volume of half a cube and then add the full cube as an additional part on the plate.

A simple example: You wish to print a sphere 1cm radius. It's a reasonable case to cut the sphere in half to print two hemispheres. Now, imagine a 1cm cube subtracted from the center of the sphere before the cut. Each hemisphere will have a void underneath 1cm X by 1cm Y by 0.5cm Z. If a 1cm cube is added to the build plate, it can be inserted between the two hemispheres to ensure they align properly.

[image]

Ideally the cube size and position could be determined automatically but a good start would be the ability to manually position and size on X,Y. With a cube only one post is required to fix the alignment of the two parts for both translation and rotation.

gege2b, 2015-02-13:
Cutting a part in half is a great feature becuse it is a kind of slicing, but doing all those other things should be done in modelling software, not in a slicer, IMHO

alexrj, 2016-12-18:
I usually do the same thing in CAD. Except I make 3mm round holes instead of square, because then I can use short chunks of 3mm filament for aligning the parts. :-)
I agree Slic3r will never have CAD capabilities, but this might be a nice to have.

alexrj, 2017-03-18:
This could be done by generating modifier meshes automatically.........