slic3r/Slic3r #2653: 【Slic3r 1.2.7-dev】Blend and antializing for the preview window

a4jp-com, 2015-02-11:
Can you turn on the blend and antializing for the 3D preview window, or can I do it in Windows through my Nvidia card's settings area?

[image]

Almost the same but mucked around with the colors in the program.
[image]

alexrj, 2015-02-11:
I did everything I could. If you have a patch I'll review it.

closed 2015-02-11 by alexrj

a4jp-com, 2015-02-11:
Here is a page with a sample of multi sampling:

http://stackoverflow.com/questions/27953949/make-opengl-polygon-edges-smooth

And another page with more details and better images:

http://www.learnopengl.com/#!Advanced-OpenGL/Anti-Aliasing