slic3r/Slic3r #2655: Deprecated "M107" Gcode for stopping fan

gege2b, 2015-02-12:
Hi all

I've noticed that when auto cooling is set, Slic3r puts a M107 Gcode to the begining of the file

According to the reprap wiki about Gcode, it's a deprecated command, and M106 S0 should be used instead

http://reprap.org/wiki/G-code#M107:_Fan_Off

I don't know exactly how it could impact newer firmwares or even if it's really a problem, but generating deprecated gcode is not a good thing IMHO

lordofhyphens, 2015-02-13:
The deprecation status of a  gcode on the wiki is not particularly binding,
as there are something like 3-5 firmware projects, each with its own ideas.
I don't think that it's a problem until some joker repurposes that code to
do something else.  It is fine to also call m106 s0. Alternatively this
could indicate that perhaps each firmware  should be its own gcode target.

alexrj, 2015-02-13:
Actually, I don't know who decided that it's deprecated. Most firmwares out there do not flag that command as deprecated. I prefer to follow the documentation of the existing firmwares rather than the wiki page which may be outdated or not very binding anyway. This is why we support multiple G-code flavors anyway, since there's no single specification.

closed 2015-02-13 by alexrj