slic3r/Slic3r #2657: 1.2.6: "Contact Z distance" of the "Support material" option problem

Zuborg, 2015-02-13:
In the "0.2 (detachable)" mode a support material does not adhere to the previously laid material. This happens when some parts of a complex model hanging over parts located below and need a support. I think the base of support columns should be printed with less weakened mode of printing or even as in the "0 (soluble)" mode.

lordofhyphens, 2015-02-13:
Iirc the detachable part of support should only be on the top of that
particular pillar (the interface layer).

Could you please provide an example configuration and part model?

alexrj, 2015-02-13:
@Zuborg, can you please read guidelines and update this report with the missing things?

Zuborg, 2015-02-14:
I did more tests and found that this problem does not depend of the mode. It appears when the tilt of the base surface is greater than some angle (about 45 degrees).
[image]

Testing STL: https://drive.google.com/file/d/0BzAa6v5hnj3fUU9WX2Q1RVRBZWc/view?usp=sharing
Config: https://drive.google.com/file/d/0BzAa6v5hnj3fQ0Qxano1anA5T2c/view?usp=sharing

hyperair, 2015-02-14:
Oddly enough I'm getting the reverse -- any support material generated on top of an existing structure is now almost impossible to remove without breaking the original part.

Zuborg, 2015-02-16:
**hyperair**, try to print my model above and tell about result!

hyperair, 2015-02-16:
[image]

Looks pretty good to me. I'll try peeling the supports after the build plate has cooled.

hyperair, 2015-02-16:
[image]

Here it is after the supports have been peeled off. Left some white remains, but overall looks good. The overhangs... have a few drooping strings though. I might need to increase the number of support material interface layers.

Zuborg, 2015-02-16:
Interesting! I see a strange blob at the bottom of the right pillar. Could You provide config.ini for that print? And what filament You used?

hyperair, 2015-02-16:
Flashforge blue PLA, printed at 200°C. Config is here: https://gist.github.com/db6d2564f090b885416f

The blob at the bottom of the pillar happens whenever I attempt to print a cylinder sideways. The bottom half of the cylinder always ends up looking slightly tapered.

alexrj, 2015-02-23:
Not sure what I should do here :-)

@Zuborg, it's unclear whether you actually tried to print or you're just looking at Repetier's preview. Can you put pictures of your print?
Repetier's preview is just wrong because G-code doesn't contain enough information. Can you look at Slic3r's 3D preview instead?

Zuborg, 2015-03-25:
Generally preview of the Repetier Host isn't far of a real print. I think a problem is a generation of the bottom interface layer only above a flat horizontal surface. One solution maybe a "side interface" from support to the nearest inclined shell. Sometimes support can stick to model's shell naturally. (Green area on the first photo, support extrusion flow above 300%... But the bottom part of the support column wasted). 
[image]
[image]

hyperair, 2015-03-26:
Doesn't look like the support towers came off at the interface, though. Most of them broke a little above the base. Perhaps you don't have sufficient cooling? I see filament everywhere, and the 45° overhang didn't come out well either.

Zuborg, 2015-03-26:
First three columns (from left to right) started with success and broken by hotend when it start to waste filament of the last (fourth) column... This filament mutilated the slanted console too. But look at  highlighted narrow red areas on the first photo. Small gap under the left column enlarged under the right and completely replaces interface layer under place where should be the last support column. We can see no any traces of interface there. (red area on the bottom photo) My experiments have indicated that the result is almost independent of the cooling power. General problem is that printing result usually unpredictable if it occurs at least partially in a "free space" when the molten filament misses a solid surface.

lordofhyphens, 2017-03-19:
@Zuborg can you check with 1.3.0-dev? Screenshots from Slic3r now are very good (better than rephost).

Current -dev builds: http://dl.slic3r.org/dev (pick newest version)
Attaching files:
https://help.github.com/articles/file-attachments-on-issues-and-pull-requests/