slic3r/Slic3r #2659: "Illegal division by zero at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 327" on git

mihai9, 2015-02-14:
i get this error after  git pull to 7a51831cb4dc6b2ae56ad24400ba1fdfed064528 when trying to export g-code regardless of .stl file used
The .stl and config files used to work before update

alexrj, 2015-02-14:
Please, complete your report according to the guidelines:
https://github.com/alexrj/Slic3r/blob/master/CONTRIBUTING.md

mihai9, 2015-02-14:
this happens when pressing Export G-Code button with any stl file ex: http://www.thingiverse.com/download:70563
git version is 43f48061425b3e4c8b8e6d7ff5af2c961294c2c9

the config is :
```
# generated by Slic3r 1.2.7-dev on Sat Feb 14 20:18:25 2015
avoid_crossing_perimeters = 1
bed_shape = 0x0,230x0,230x230,0x230
bed_temperature = 110
before_layer_gcode = 
bottom_solid_layers = 2
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S20 ; Set hotend temp, no wait.\nG92 E0 ; Reset Extruder to 0.\nG1 E-5 F1800 ; Retract filament 5mm @30mm/s.\nG92 E0 ; Reset Extruder to 0. \nM113 S0\nG1 Y230 F3000 ; Bring table forward\nM140 S10 ; Set bed temp, no wait.\nM300 S659 P125 ; Play Beep on Completion. Fur Elise\nM300 S622 P125\nM300 S659 P125\nM300 S622 P125\nM300 S659 P125\nM300 S494 P125\nM300 S587 P125\nM300 S523 P125\nM300 S440 P500\nM1 ; Shut heat and motors off.\nM107 ; Turn Fan off.
external_fill_pattern = concentric
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 1
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1.05
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1.73
fill_angle = 45
fill_density = 20%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_bed_temperature = 120
first_layer_extrusion_width = 120%
first_layer_height = 0.2
```
the generated file contains only

```
; generated by Slic3r 1.2.7-dev on 2015-02-14 at 20:23:34

; external perimeters extrusion width = 0.40mm
; perimeters extrusion width = 0.67mm
; infill extrusion width = 0.67mm
; solid infill extrusion width = 0.67mm
; top infill extrusion width = 0.67mm

M107 ; disable fan
M190 S30 ; Start bed heating. Wait for temp
M140 S110 ; Set bed temp, no wait.
M104 S240 ; Set hotend temp, no wait.
G28 ; home all axes
G1 Z10 F5000 ; lift nozzle
M140 S110 ; Set bed temp, no wait.
M109 S120 ; Set hotend temp, wait.
M106 S80; Begin Fan to cool entry tube and stop swell & jamming.
M190 S110 ; Start bed heating. Wait for temp.
M109 S245 ; Set hotend temp, wait.
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
```

[image]

mihai9, 2015-02-15:
it works for an 1milimeter cube but not for an 1 centimeter cube

mihai9, 2015-02-16:
mabe this helps more:
```
PERL5LIB=~/perl5/lib/perl5/  perl -MCarp::Always ./slic3r.pl
```
```
Illegal division by zero at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 327 thread 16.
	Slic3r::GCode::_extrude_path(Slic3r::GCode=HASH(0x64c98e0), Slic3r::ExtrusionPath::Ref=SCALAR(0x7f00ac0784a0), "infill", undef) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 235 thread 16
	Slic3r::GCode::extrude_path(Slic3r::GCode=HASH(0x64c98e0), Slic3r::ExtrusionPath::Ref=SCALAR(0x7f00ac0784a0), "infill") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 109 thread 16
	Slic3r::GCode::extrude(Slic3r::GCode=HASH(0x64c98e0), Slic3r::ExtrusionPath::Ref=SCALAR(0x7f00ac0784a0), "infill") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print/GCode.pm line 547 thread 16
	Slic3r::Print::GCode::_extrude_infill(Slic3r::Print::GCode=HASH(0x693f468), HASH(0x7f00ac094d48)) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print/GCode.pm line 496 thread 16
	Slic3r::Print::GCode::process_layer(Slic3r::Print::GCode=HASH(0x693f468), Slic3r::Layer::Ref=SCALAR(0x7f00ac07ac08), ARRAY(0x7f00ac07ac50)) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print/GCode.pm line 240 thread 16
	Slic3r::Print::GCode::export(Slic3r::Print::GCode=HASH(0x693f468)) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print.pm line 413 thread 16
	Slic3r::Print::write_gcode(Slic3r::Print=SCALAR(0x6ee2760), "/home/u1/lucru/3d/30X20X10.gcode") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print.pm line 84 thread 16
	Slic3r::Print::export_gcode(Slic3r::Print=SCALAR(0x6ee2760), "output_file", "/home/u1/lucru/3d/30X20X10.gcode") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GUI/Plater.pm line 1100 thread 16
	eval {...} called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GUI/Plater.pm line 1099 thread 16
	Slic3r::GUI::Plater::__ANON__() called at /home/u1/lucru/3d/Slic3r/lib/Slic3r.pm line 115 thread 16
	Slic3r::__ANON__() called at /home/u1/lucru/3d/Slic3r/lib/Slic3r.pm line 116 thread 16
	eval {...} called at /home/u1/lucru/3d/Slic3r/lib/Slic3r.pm line 116 thread 16
```

mihai9, 2015-02-16:
this happends when autocooling is enabled
also setiing pressure advance in printer settings gives :
Use of uninitialized value in multiplication (*) Slic3r/lib/Slic3r/GCode/PressureRegulator.pm line 46

alexrj, 2015-02-16:
Why did you close this?

alexrj, 2015-02-17:
No answer fro @mihai9 :-(

lonelymyp, 2015-03-14:
I have the same problem in version 1.2.2 and 1.2.6 independent of the model when trying to make a raft. 
If the raft layers is greater than 1.
18:58:34.879 : <Slic3r> Illegal division by zero at D:/3DPRIN~1/SLIC3R~1.6/lib/std/Slic3r/GCode.pm line 327.

generated by Slic3r 1.2.6 on Sat Mar 14 19:14:43 2015
avoid_crossing_perimeters = 1
bed_shape = 0x0,200x0,200x200,0x200
bed_temperature = 120
before_layer_gcode = 
bottom_solid_layers = 2
bridge_acceleration = 1000
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 65
brim_width = 3
complete_objects = 0
cooling = 1
default_acceleration = 1000
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0.5
external_perimeter_speed = 40
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_diameter = 1.75
fill_angle = 0
fill_density = 10%
fill_pattern = rectilinear
first_layer_acceleration = 200
first_layer_bed_temperature = 120
first_layer_extrusion_width = 200%
first_layer_height = 0.4
first_layer_speed = 5
first_layer_temperature = 250
gap_fill_speed = 10
gcode_arcs = 0
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 1000
infill_every_layers = 2
infill_extruder = 1
infill_extrusion_width = 120%
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 20%
infill_speed = 60
interface_shells = 0
layer_gcode = 
layer_height = 0.24
max_fan_speed = 100
min_fan_speed = 35
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.5
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 200
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 60
perimeters = 3
post_process = 
pressure_advance = 0
raft_layers = 3
resolution = 0
retract_before_travel = 2
retract_layer_change = 0
retract_length = 6
retract_length_toolchange = 10
retract_lift = 0.3
retract_restart_extra = -0.2
retract_restart_extra_toolchange = 0
retract_speed = 150
seam_position = aligned
skirt_distance = 0
skirt_height = 2
skirts = 0
slowdown_below_layer_time = 15
small_perimeter_speed = 40
solid_infill_below_area = 5
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 90%
solid_infill_speed = 60
spiral_vase = 0
standby_temperature_delta = -5
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 1
support_material_angle = 1
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 2
support_material_interface_speed = 0
support_material_pattern = rectilinear-grid
support_material_spacing = 2.5
support_material_speed = 20
support_material_threshold = 0
temperature = 240
thin_walls = 1
threads = 4
toolchange_gcode = 
top_infill_extrusion_width = 90%
top_solid_infill_speed = 60
top_solid_layers = 3
travel_speed = 100
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 0
xy_size_compensation = 0
z_offset = 0

mihai9, 2015-04-23:
i'm sorry for the long delay, but i've been busy (and hoping it will get fixed anyway ;-( )
now i'm back and found a little more about it
the issue is still present in 5eb3bc52efa807c0439dba1ec0971fc37850d51e, it appeares in my case when using a specific print settings file on any object that has infill generated (this explains why the 1 milimeter cube is working) and auto cooling is checked.
i did not attempted to generate raft layers
i came with this file because i was using a config file that came with the printer and playing with various settings
starting with the defaults and editing settings results in a configuration that works

any ideea on how to identify the specific parameter ?
anything i changed from the interface so far did not affected the issue

to me it looks that Slic3r::GCode::_extrude_path is called with an undefined parameter (speed) because Slic3r::GCode::extrude_path is called with only three parameters but then i'm lost

this is the offending file:
```
# generated by Slic3r 1.2.7-dev on Sun Mar 22 13:46:02 2015
avoid_crossing_perimeters = 1
bottom_solid_layers = 2
bridge_acceleration = 0
bridge_flow_ratio = 0.95
bridge_speed = 60
brim_width = 0
complete_objects = 0
default_acceleration = 0
dont_support_bridges = 1
external_fill_pattern = concentric
external_perimeter_extrusion_width = 150%
external_perimeter_speed = 50%
external_perimeters_first = 1
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0
fill_angle = 45
fill_density = 100%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_extrusion_width = 0
first_layer_height = 0.2
first_layer_speed = 25
gap_fill_speed = 20
gcode_comments = 1
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 35
interface_shells = 1
layer_height = 0.2
min_skirt_length = 5
notes = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 32
perimeters = 2
post_process = 
raft_layers = 0
resolution = 0
seam_position = random
skirt_distance = 6
skirt_height = 1
skirts = 4
small_perimeter_speed = 15
solid_infill_below_area = 10
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 0
spiral_vase = 0
standby_temperature_delta = -5
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
thin_walls = 1
threads = 6
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 2
travel_speed = 100
xy_size_compensation = 0
```

trace from the current version (5eb3bc52efa807c0439dba1ec0971fc37850d51e):
```
Illegal division by zero at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 327 thread 43.
	Slic3r::GCode::_extrude_path(Slic3r::GCode=HASH(0x7fa788065248), Slic3r::ExtrusionPath::Ref=SCALAR(0x7fa7883feff8), "infill", undef) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 235 thread 43
	Slic3r::GCode::extrude_path(Slic3r::GCode=HASH(0x7fa788065248), Slic3r::ExtrusionPath::Ref=SCALAR(0x7fa7883feff8), "infill") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GCode.pm line 109 thread 43
	Slic3r::GCode::extrude(Slic3r::GCode=HASH(0x7fa788065248), Slic3r::ExtrusionPath::Ref=SCALAR(0x7fa7883feff8), "infill") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print/GCode.pm line 555 thread 43
	Slic3r::Print::GCode::_extrude_infill(Slic3r::Print::GCode=HASH(0xab1c7f0), HASH(0x7fa7883fcad0)) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print/GCode.pm line 508 thread 43
	Slic3r::Print::GCode::process_layer(Slic3r::Print::GCode=HASH(0xab1c7f0), Slic3r::Layer::Ref=SCALAR(0x7fa7883f3a70), ARRAY(0x7fa7883f3ab8)) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print/GCode.pm line 247 thread 43
	Slic3r::Print::GCode::export(Slic3r::Print::GCode=HASH(0xab1c7f0)) called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print.pm line 412 thread 43
	Slic3r::Print::write_gcode(Slic3r::Print=SCALAR(0xa126230), "/home/u1/lucru/3d/New folder (2)/penholder_v1.gcode") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/Print.pm line 84 thread 43
	Slic3r::Print::export_gcode(Slic3r::Print=SCALAR(0xa126230), "output_file", "/home/u1/lucru/3d/New folder (2)/penholder_v1.gcode") called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GUI/Plater.pm line 1112 thread 43
	eval {...} called at /home/u1/lucru/3d/Slic3r/lib/Slic3r/GUI/Plater.pm line 1111 thread 43
	Slic3r::GUI::Plater::__ANON__() called at /home/u1/lucru/3d/Slic3r/lib/Slic3r.pm line 113 thread 43
	Slic3r::__ANON__() called at /home/u1/lucru/3d/Slic3r/lib/Slic3r.pm line 114 thread 43
	eval {...} called at /home/u1/lucru/3d/Slic3r/lib/Slic3r.pm line 114 thread 43
```

alexrj, 2015-06-02:
@mihai9, I tested your last config with a 20mm cube and I got no crash so this looks fixed.  Let me know if this is not the case. Thank you for reporting the issue.

closed 2015-06-02 by alexrj

pcarew, 2017-05-04:
I had this issue on Ver 1.1.7
I resolved it by changing 'First Layer Speed" I had inadvertently set it to '0'