slic3r/Slic3r #2660: Slic3r 1.2.6 stopt when loadingbar is allmost @ 100%

spirit18, 2015-02-15:
first of all i like your software
but the latest beta version stops working if the loadingbar is allmost at 100%
look movie i added.
https://www.dropbox.com/s/s9e4682clr40sco/Desktop%2002.15.2015%20-%2010.35.46.01.mp4?dl=0

alexrj, 2015-02-15:
Can you please provide the things described in the guidelines please? Otherwise it's impossible to reproduce the problem.

spirit18, 2015-02-15:
Oww sorry my bad never seen the guidelines.
Config file : https://www.dropbox.com/s/63wwwylgepdx9vk/%232660.ini?dl=0
Object: https://www.dropbox.com/s/8jefcrjqlxy1dqo/feeder%20hulp%20v1%20flipt%20180.stl?dl=0

PC specs:
Windows 7 Home premium (64bit)
AMD FX-6100 (@ 6x 3.6Ghz)
8 GbRam
2x 256 SSD raid0
Nvidia GTX 770, 4GbRam
ROG crosshair V formula-Z, motherboard.

Steps to reproduce:
1) open Slic3r 1.2.6 (32 or 64 bit)
2) add object
3) wait until loading bar is @ 100%
4) Slic3r stops working.
There are NO error massages

justmyopinion, 2015-02-15:
Maybe only lack of patience here.
I have about the same setup and do slice your file, but it may take some more time than expected.
Reason is a known lack of performance with Slic3r when "printsettings/avoid_crossing_perimeters" has been enabled, causing slicetime to be excessive.
Uncheck this and see that your slice will be fine.

spirit18, 2015-02-15:
i dont mind waiting sometimes it take up to 30 min.
but the problem as you can see on the Movie (seems you did not look at it, more a lack of patience to read the whole issue report here.)
Slic3r just Closes when the Loading bar is near 100%

justmyopinion, 2015-02-15:
ok, but that is not what happens with ver 1.2.6, my Win7/64 and your config/STL file. It slices fine in both cases also when exporting gcode.

spirit18, 2015-02-15:
Oke nice.
i found out the issue i have. 

Poeple (like me) need to update Windows 7. (realy odd as i got new udates yesterday) 
and i think its because i did not have the latest "microsoft .NET Framework 4.5.2" now it work fine :-D

some one can close this issue.

justmyopinion, 2015-02-15:
Well the issue was not described as "crashing, but just "stopped working" :-) so for security reason I mostly avoid entering movies.
Never mind, good you found a solution.

spirit18, 2015-02-15:
it depant on how you look at "crashing" and "stopped wroking"
but i see your point.
"crashing" to me, means i get a massege from windows asking me if i like to "End Task " (Not responding) so the software realy crashes.
"stops working" means No message, just pops back to desktop.

spirit18, 2015-02-15:
oke wait its back again.. strange!

lordofhyphens, 2015-02-15:
We need to know what settings you were using (export config) and a copy of
a model you were seeing this issue with. Otherwise we can't reproduce the
issue. If we can't reproduce, can't fix.

SpencerCWatson, 2015-02-17:
Hopefully I can add to fixing this because I am having the same issue. I happens in both windows 7(My other computer) and 8.1.

 It happens just as the video shows, add part and right before the loading bar is full Slic3r will close. For me it mostly happens with round objects and it does not matter if it is my settings or default settings.

PC specs:
Windows 8.1 (64bit)
Intel i7 4 core 2.8GHz
8 GbRam
256 SSD
Nvidia GTX 870M

https://drive.google.com/folderview?id=0BwItvBIsxpO9blo4OWlzRnNFT28&usp=sharing

SpencerCWatson, 2015-02-17:
OK I have been playing with this for an hour now and it looks like deleting the C:\Users\(your own personal user name)\AppData\Roaming\Slic3r and then importing my settings back in fixed it. Not sure why I had issues in the first place but it seems to be working now.

alexrj, 2015-06-02:
@spirit18 and @SpencerCWatson, this looks fixed as of 1.2.7.  Please feel free to reopen if not. Thank you for reporting the issue.

closed 2015-06-02 by alexrj