slic3r/Slic3r #2661: Use medial axis/gap fill logic for 100% concentric infill too

alexrj, 2015-02-15:
The outer wall is a 100% density concentric infill but has a gap inside:

[image]

jonaskuehling, 2015-03-09:
Ah, good, already covered...

Do you have a milestone for this already? I'm starting to love 100% concentric infill these days ;-)

kefir-, 2015-05-27:
Is this what I'm seeing here? I've set the top/bottom infill to concentric, and then I got some spaces at the center of the infill. The one at the bottom was the worst:
[image]
Scaling up and down, it got better or worse. If I were to guess, it seems like the center extrusion has to fit exactly, otherwise it's skipped.