slic3r/Slic3r #2662: 1.2.6 experimental: First layer extrusion width doesn't affect first raft-/support-layer

jonaskuehling, 2015-02-16:
Hi Alessandro,

not sure if this is intended behaviour, but I just noticed that changing the first layer extrusion width (like 150% or 200%...) doesn't affect the first layer of support (raft), I mean those lines with gaps, first layer on print bed. The first layer height is used as configured, though.

Maybe I just don't see the good reason for it, why Slic3r prefers to always calculate the extrusion width on its own instead of using the pontentially modified extrusion width from advanced settings. But beeing able to do further modification here seems quite interesting to me, since one could force even fatter traces in the first layer to increase bed leveling redundancy and prind bed adhesion for certain support materials.

GCODE/CONFIG: https://www.dropbox.com/s/0lm0vojyhu6clz9/spool-wings_750g_2300g.gcode?dl=0

jonaskuehling, 2015-02-16:
Btw., brim is generated with the configured extrusion width, but the support base line pattern ain't.

hyperair, 2015-02-17:
I get this too. I had a look around inside the source code, and it doesn't look like it's intentional. I haven't figured out why it's not honouring the first layer extrusion width though.

hyperair, 2015-02-17:
Aha, found it:
```diff
--- lib/Slic3r/Print/Object.pm  2015-02-02 02:32:41.882955504 +0800
+++ /usr/share/perl5/Slic3r/Print/Object.pm     2015-02-17 13:18:08.016468134 +0800
@@ -547,14 +547,14 @@
     my ($self) = @_;
     
     my $first_layer_flow = Slic3r::Flow->new_from_width(
-        width               => ($self->config->first_layer_extrusion_width || $self->config->support_material_extrusion_width),
+        width               => ($self->print->config->first_layer_extrusion_width || $self->config->support_material_extrusion_width),
         role                => FLOW_ROLE_SUPPORT_MATERIAL,
         nozzle_diameter     => $self->print->config->nozzle_diameter->[ $self->config->support_material_extruder-1 ]
                                 // $self->print->config->nozzle_diameter->[0],
         layer_height        => $self->config->get_abs_value('first_layer_height'),
         bridge_flow_ratio   => 0,
     );
```

Looks like first_layer_extrusion_width moved from PrintObjectConfig to PrintConfig, so we need to be looking at $self->print->config instead of $self->config for first layer extrusion width.

alexrj, 2015-02-22:
Thank you for the patch @hyperair! Much appreciated.

jonaskuehling, 2015-02-27:
Hi,

great, at least first layer extrusion width is now respected for support's first layer at all :-) Unfortunately the line spacing isn't adjusted accordingly but stays the same as with unmodified first layer extrusion width, leading to heavily overstuffed first layer with wider extrusion width configured. I tested two settings for this case:

First layer extrusion width = 100% (looks like intended so far)
[image]

First layer extrusion width = 500% (line spacing same as with 100% but correctly wider traces. Unfortunately no gaps left..)
[image]

Looks like the line spacing is somehow hard-coded at the moment or at least not dependend on the extrusion width but layer height only?

alexrj, 2015-02-27:
Good catch @jonaskuehling, I should have fixed it now. Let me know!

jonaskuehling, 2015-02-27:
Hell yeah! Good work @alexrj !

[image]

alexrj, 2015-02-27:
Good then!

closed 2015-02-27 by alexrj

a4jp-com, 2015-02-28:
Very nice fix ^^

PR #2667 (2015-02-22): Fix support material first layer extrusion width
This fixes an issue where the support material doesn't honour a custom first
layer extrusion width.

Fixes: #2662
commit 073681a5 (2015-02-22): Get first_layer_extrusion_width from PrintConfig instead of PrintObjectConfig

This fixes an issue where the support material doesn't honour a custom first
layer extrusion width.

Fixes: #2662