slic3r/Slic3r #2663: Feature request: offer preview for existing GCODE

jonaskuehling, 2015-02-16:
Hi,

I'd love to see a functionality to open and preview existing GCODE files with the new 3D awesomeness!

Maybe this could be offered in a separate window via the app menu, making use of the "preview" and "layers" tabs like they are available in the main slic3r/plater window..?

Shouldn't necessarily be integrated directly into the Slic3r window, since this is a pure static GCODE prewiev functionality and has not so much to do with slicing anymore. See it as a tool that just opens the choose file dialogue and then opens a standalone windows just showing the loaded file name and the two mentioned tabs with 3D and 2D GCODE visualisation.

Would be incredibly usefull for later gcode comparison when something should be printed again with modified parameters, or general later GCODE diagnosis.

Of course this could require the "verbose gcode" option being enabled, to have access to e.g. the used extrusion widths of the loaded GCODE. But as this option is totally usefull anyway, it wouldn't hurt if one would need to enable it first to make use of the preview tool....

alexrj, 2015-02-22:
I entirely agree that it would be nice. :-)
It would need different and new code, as the current preview reads data from internal geometric structures and not from G-code-like instructions. But the main problem is that even verbose G-code wouldn't help: we'd need to dump a lot more things into the file to display the file correctly, like vertical spans (for "Combine infill every 'n' perimeters" and support material layers height…)

gege2b, 2015-02-23:
Hi

If the goal is simply to have a basic preview, maybe we could get rid of support layer heigt and so, and just tracing lines with the same weight ? I.E. as pronterface does (it's the only host software I use)
well, maybe its a dumb question sorry :)

jonaskuehling, 2015-02-23:
Ok, I see.. Feared this kind of answer, @alexrj !

So, without priority at all: A true printer simulation GCODE visualisation would probably be amazing ;-) So "just" some rendering application that virtually prints the actual GCODE, based on the given movements and extrusion quantities... Ok, maybe volumetric extrusion would be quite handy here, but I already read about it somewhere relating to Slic3r?

This kind of visualisation would definitely highlight all bugs that may make it into GCODE during slicing I think. Disclaimer: Didn't think about this too intensely, so I may miss this big highly visible reason why nobody did it yet ;)
(Or is it basically because of the volumetric extrusion thing that'd make it much easier?)

lordofhyphens, 2015-02-23:
Why are you loading gcode from other sources into slic3r in the first place?

gege2b, 2015-02-23:
I guess he's not loading anything, as this is a feature request to be able to preview "external" gcode into slic3r

lordofhyphens, 2017-03-19:
There's some SVG output work done by @bubnikv that may be better for debug purposes.