slic3r/Slic3r #2664: [1.2.7-dev] Extra perimeter regression

Vicious-one, 2015-02-16:
Now it produces this result on top layer when extra perimeters are enabled:
(2 perimeters + extra)
[image]

alexrj, 2015-02-23:
Thank you, fixed! I hope is much more robust now.

closed 2015-02-23 by alexrj

Vicious-one, 2015-02-24:
Seems fixed for this model, thanks, but "30%" logic will most probably fail on some other tricky case.

alexrj, 2015-02-24:
Let's see… the threshold itself might change (40%? 50%?) but I think the logic is robust… I couldn't find any other geometric definition of extra perimeters. They're very very tricky to *define*

Vicious-one, 2015-02-24:
IMO, the term itself is a bit ambiguous. 

This feature, as I understand it, appeared as an attempt to cover the unsightly infill overlap/overextrusion so that it won't be visible from above.

If I were the one implementing the logic, I would have done it like that:
0) Assume there's a layer below the current one which has P perimeters.
1) Generate an external perimeter for a current layer
2) Walk the entire generated perimeter in small steps (like 1 mm) and check if there's an infill region directly below each point.
3) If the quantity of points which have infill region below them is greater than N (N > 0), recalculate the layer below with P=P+1 perimeters.
4) Goto step 2 and repeat this untill the infill region appears under less than N points of the external perimeter of the current layer.

alexrj, 2015-02-24:
Tuning your N or the area delta threshold is the same as my 30% :-)

The algorithmic problem is that your upper layer might be much smaller or it might have a very different shape from the lower one, but in case a part of its contour would benefit from an extra lower perimeter we need to check whether it's worth or not.  Adding one or more extra perimeters just for hiding a very small quantity of infill endpoints it not worth.  This is why the current logic only does that if the two layers share at least 30% of their shape… Maybe this 30% should even become 50%, but I'll wait for test cases...

Vicious-one, 2015-02-24:
Anyway, in case this actually backfires, there can be a workaround: use a maximum extra perimeters threshold.

curiouspl2, 2015-05-25:
once again - You did introduce very nice variable, but You are somehow reluctant to put it into EXPERT options :) it would fit VERY nicely on right side of the 'extra perimeters if needed' as simply 'threshold' with some nice hint :)
I think the setting will greatly depend on layer height users try to print with and how badly they need their model being watertight from the top (in case of models which are subjected to weathering outside - rain, snow, etc - this is actually usefull) or how much plastic they want to save .

also for very big, complex, or 'production' models this variable could be used like many other from modifier options , allowing to set different thresholds for different parts of model...