slic3r/Slic3r #2665: Perspective View in Preview and 3D View

callimero, 2015-02-16:
Hi,

I really miss a perspective view in 3D and Preview. Of course the ortho view has its pros, I think we could use a perspective view for a better look at large comlpicated parts.

Carsten

a4jp-com, 2015-02-16:
Some models look weird in ortho mode. I would also like a perspective view or switch/setting to choose if possible.

LeChuck71, 2015-02-23:
Hi,

in all my CAD packages I turn on perspective, ortho is just a pain in the eyes.
Please make a "Perspective 3D Window" option.

Cheers
Axel

a4jp-com, 2015-02-23:
Is there a reason why ortho is good or is it difficult to make?

A perspective view seems so much more natural for 3D designers and that is the way our eyes naturally see objects. Even just making the window a tiny bit perspective would be nice if possible.

LeChuck71, 2015-09-18:
Any news on this "minor" issue?

ghuser2, 2015-09-20:
I would imaging that perspective would require a lot more processing, but I don't know for sure.

phizz166, 2015-10-05:
Rendering in perspective doesn't require any more processing power than isometric. Consider that an isometric view is equivalent to a perspective view where the virtual camera has infinite focal length. It's a different set of transformations, but not ones that are significantly more complex.

Isometric is good where you need to compare the sizes of multiple objects. With no foreshortening, you know that e.g. 10cm measured anywhere in the drawing always represents 10cm in reality. This can be very useful in CAD drafting and modeling, but I can't think of how it's helpful in Slic3r.

+1 to the idea of a perspective toggle.

Nprod, 2016-03-19:
+1 from me on that too, Perspective is no different from Ortho, it's more a matter of preference. If anything, it's easier to navigate in Perspective.

pixsrv, 2016-05-17:
I also vote for perspective view

bubnikv, 2016-05-17:
Perspective view should be trivial to implement, but I am not quite sure how useful it would be. In the case of the perspective / ortho camera toggle, one shall weight the usefulness of the feature against the  user interface clutter.

lordofhyphens, 2016-05-17:
Yeah, I can't really see the point of perspective view. This isn't a
modeling/art program.

pixsrv, 2016-05-17:
@lordofhyphens AFAIK many people work on perspective view only, including me. I never use ortho view in any 3d app as this view is not natural in any case and for me this it's 100% useless.

Nprod, 2016-05-17:
@lordofhyphens Just because you don't use it doesn't make it pointless... Perspective is used to get a better perception of depth and scale. You can't inspect an object up close in Ortho if it's obscured by something else because it has no depth. It's funny you say "this is not a modelling program" because Ortho is used a lot in modelling to ensure dimensional accuracy without perspective distorsion.

lordofhyphens, 2016-05-17:
The larger point here is that development time is limited. So features only
really get added if they are interesting to do or affect our use case (that
is, we need it). Bugs get their own priority class.

If you really want it in the interface soon, code it (or pay someone else
to do so) and make a pull request. With regards to interface, I don't see a
real argument against if the option to change it is in the Preferences menu
(like the simple/expert settings). It's not really something that should be
expected to change often (unlike a general modeling/art program, where you
want to be able to preview the scene) Odds are the actual changes to set it
to Perspective live in this file:
https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/3DScene.pm

It's Perl, so once you get the bulk of it compiled, you can play around
with the GL function calls in Perl and only have to close/reopen the
program. Adding the option (and making it respected) to the menu would
probably take a little more digging, but I am pretty sure it's all Perl
still.

If I see a pull request that doesn't break anything and adds this feature;
I would merge it in.

a4jp-com, 2016-05-17:
Nice.

pixsrv, 2016-05-17:
@lordofhyphens no doubt that bugs are the priority. But new features especially such fundamental are also very welcome for sure. I tell you when I use slic3r I see the lack of perspective view all the time but bugs (despite they stop you) appear accidentially ;)
You can believe me or not but I've never meet any bug in slic3r on my own.

lordofhyphens, 2016-05-17:
@pixsrv and I've never wished for perspective view on the plater :). I
have, however, seen the error in gapfill logic firsthand.

Near as I can tell permanently changing to perspective mode looks pretty
straightforward to hack together from my notes above.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Nprod, 2016-05-17:
@lordofhyphens If this is a matter of limited development time then yes, bugs always get top priority over feature requests. Having a Persp/Ortho setting in the Preferences menu is enough, i think people are going to use only one of those modes anyway.

singh1114, 2017-04-05:
@alexrj, 

According to the Wikipedia, the main difference in perspective and the `orthographic view is the position of the camera. In orthographic the camera is at infinity while in perspective the camera is at some point. Is it the basics of having the perspective view?

If yes, then what should be the default position of the camera and what files should I look into to solve the issue?

lordofhyphens, 2017-04-05:
Grep is your friend. Google the different constants/functions  used by
opengl.

Grep for them in the source directory. (-r is a useful flag here).

lordofhyphens, 2017-04-11:
@singh1114 everything related to GL transforms and camera position are in 3DScene.pm, pay close attention to Render() and Resize(). 

Use your own best judgement as to where to put the camera.

lordofhyphens, 2017-04-11:
@singh1114 also consider http://www.glprogramming.com/red/chapter03.html and use of either gluPerspective or glFrustum.

Have fun messing with the near/far buffers too. 

The following diff will add a new option to the Preferences dialog, which you can reference in 3DScene.pm with "$Slic3r::GUI::Settings->{_}{view_mode}"
````
diff --git a/lib/Slic3r/GUI.pm b/lib/Slic3r/GUI.pm
index ce94d08..adae7dc 100644
--- a/lib/Slic3r/GUI.pm
+++ b/lib/Slic3r/GUI.pm
@@ -87,6 +87,7 @@ our $Settings = {
         no_controller => 0,
         threads => $Slic3r::Config::Options->{threads}{default},
         color_toolpaths_by => 'role',
+        view_mode => 'ortho',
     },
 };
 
diff --git a/lib/Slic3r/GUI/Preferences.pm b/lib/Slic3r/GUI/Preferences.pm
index 9c96efa..a4ca38b 100644
--- a/lib/Slic3r/GUI/Preferences.pm
+++ b/lib/Slic3r/GUI/Preferences.pm
@@ -71,6 +71,16 @@ sub new {
         tooltip     => 'Disable communication with the printer over a serial / USB cable. This simplifies the user interface in case the printer is never attached to the computer.',
         default     => $Slic3r::GUI::Settings->{_}{no_controller},
     ));
+    $optgroup->append_single_option_line(Slic3r::GUI::OptionsGroup::Option->new(
+        opt_id      => 'view_mode',
+        type        => 'select',
+        label       => '3D View Mode',
+        tooltip     => 'Choose between orthographic view in 3D (default) and a less accurate perspective view (artsy).',
+        labels      => ['Orthographic','Perspective'],
+        values      => ['ortho','perspective'],
+        default     => $Slic3r::GUI::Settings->{_}{view_mode},
+        width       => 130,
+    ));
     
     my $sizer = Wx::BoxSizer->new(wxVERTICAL);
     $sizer->Add($optgroup->sizer, 0, wxEXPAND | wxBOTTOM | wxLEFT | wxRIGHT, 10);
````

Serveurperso, 2019-06-16:
What news ? This crap isometric view drives me crazy !!!!! Please enable perspective by default ! We don't need a GUI configuration option because isometric is totally useless for a non-CAD preview.

a4jp-com, 2019-06-16:
I had forgotten I wrote in here. If it's still okay, I'd like to see models in a perspective view instead of orthographic view too.

Nprod, 2019-06-17:
According to Prusa himself, perspective view should be coming in later versions of his PrusaSlicer fork, the GUI has already had some very nice improvements there.

callimero, 2019-06-18:
I can't believe its more than 4 years now :)

As I use the prusa fork only since. I am temped to close this just not to beat a dead horse.

callimero, 2021-05-12:
Bye! And thanks for all the fish!

closed 2021-05-12 by callimero

a4jp-com, 2021-06-11:
Did this ever get added? I started using other programs but I'd love to see what has been happening here again.

pixsrv, 2021-06-11:
i'm using prusa slicer and it has perspective view
BUT!!!
as far as i can tell it's perspective with ortho faces
which means it' still strange

it's really strange for me how it's possible to have such a great product on one hand
with such lack of fundamental feature

it's like having a graphics app but only in grayscale

supermerill, 2021-06-11:
> as far as i can tell it's perspective with ortho faces

?

pixsrv, 2021-06-11:
regular perspective view has 3 points where parallel lines are crossing eachother
in persp with ortho faces there is less of them
it can be 2 or even just one

a4jp-com, 2021-06-11:
I can see version 1.3.0 (2018-05-10) is on the website (https://slic3r.org/releases/1.3.0/). Has much changed since then?