slic3r/Slic3r #2666: Feature request: Save to Printer As....

callimero, 2015-02-16:
Often I would like to have a new Filename send to the printer. e.g. I have send a clip.gcode to the printer (Octoprint in my case) I would not overwrite it with a new version (changed slicing params or want to print it in an array etc,). Then a Send to Printer As... which requests a new name would be handy.

lordofhyphens, 2015-02-17:
I think this can be written in through downloading http://$octoprint_server/api/files, parsing the output for the gcode names and checking to see if the current one is on there and popping up a confirmation box (with an edit option pre-filled in for the same value) if there's a match. Adds a click, but you could change the name that way without making things much more complex.

callimero, 2015-02-17:
I was thinking to make a normal Send to Printer when clicking the Button with LMB. But make it Send To Printer As... when holding the SHIFT Key while clicking.

lordofhyphens, 2015-02-17:
... that is extremely obtuse as an option on a button and violates multiple
assumptions regarding buttons. I think it would be better to leave a config
option in the octoprint area to select between overwrite or set custom name
on upload  (prefilled to the default value of what it would be in print
settings)

A workaround that I just considered is to add a date/time variable flag (a
var for some of the print settings used) to the gcode output filename.

alexrj, 2015-02-17:
Does Octoprint really overwrite files with same name?

callimero, 2015-02-17:
@alexrj yes, and in 90% of the cases it is wanted. We all (most?) print an object in the design and finding the right slicing params phase more than one time. But then there are the 10% were I want a different file to test something or to print a part multiple times. So for now I need to save the gcode then upload it by hand to octoprint. 

I don't want to taken as lazy but these "pro" features are ones which makes working with a software streamlined. I was hoping it is a usefull and easy to implement feature. Maybe not so wanted (by me ;-)) as the automatic or manual reload of a 3D (STL) model but still...

lordofhyphens, 2015-02-17:
The solution is straightforward,  it just requires a parser for the format
that octoprint uses.

The UI choices to make it happen are harder.

starno, 2015-02-24:
I was able to work around this issue for the most part, but the current functionality doesn't make any sense in the situation of using slic3r to create printing plates of multiple parts within the same build.  You either have to export it as STL just to be able to name the file, then re-import it and slic3 it again or just rely on future self remembering the name doesn't match the contents.

A potential solution that wont clutter the UI or add more popup windows is to utilize the space in the large white box that lists the contents of the current slic3, something similar to this:
[image]
The content tree arrangement is intuitive imo.  The filename can default to anything, as long as it is a rename-able field and the send to printer simplicity is not sacrificed.

furanshishouco, 2015-03-02:
And a textbox below "Printer settings" where write a name for the file? Preloaded with added stl's name.

pyrophreek, 2015-07-09:
I would also like to see this functionality added.  The last suggestion, the textbox, sounds like the most streamlined and clean option.  Name it "Output File Name" or something similar.  This would be great for both plating parts (currently only named by the first stl, but sometimes I want to print just the first part as a replacement or extra, and keep the plated arrangement available) and for printing the same model with different settings, such as different filament colors or layer heights.

alexrj, 2015-07-09:
**Pro tip:** under Print Settings there's an option named Output Filename Format. You can set that to `[input_filename_base]_[timestamp].gcode` and you'll have unique filenames automagically.

I like the first idea by @lordofhyphens (query the OctoPrint host and prompt user in case the object already exists). Maybe we should always prompt user when printing a multi-object plate (case reported by @starno).

alexrj, 2016-12-17:
Okay, Slic3r now prompts with the filename and asks whether to start printing. If a file with the same name exists, user is asked for confirmation.

closed 2016-12-17 by alexrj