slic3r/Slic3r #2670: Post-Processing script works in command line but not in GUI

Hertafeld, 2015-02-17:
Hi all,
I've written a simple post-processing expect script to upload gcode files to remote server:

    #!/usr/bin/expect

    lassign $argv arg1
    spawn scp $arg1 "MY_REMOTE_LOCATION"
    #######################
    expect {
      -re ".*es.*o.*" {
        exp_send "yes\r"
        exp_continue
      }
      -re ".*sword.*" {
        exp_send "MY_REMOTE_PASSWORD\r"
      }
    }
    interact

It executes just fine when I run this program from the command line with a single argument. It also runs properly as a postprocess when I run slic3r from the command line:

    /Applications/Slic3r.app/Contents/MacOS/slic3r /path/to/stl --post-process /path/to/script

But when I run it from the GUI, there is a pause during slice as it claims to execute my post-process, and the file is not uploaded. Read and write permission is enabled for all. I have a single line, the absolute path to my script, in my output options in the GUI.

Is there a difference between the way the command line slic3r and the GUI handle these scripts? Is there a way to see any exceptions or output from my script when it's run from the command line so I can debug? What's going wrong here?

I'm running OSX 10.10.1 and my remote location is running Raspbian. I'm using Slic3r 1.1.7.

I much appreciate any guidance.

-Evan

alexrj, 2015-02-22:
There shouldn't be any difference except for relative paths but that's not your case… Maybe Console.app shows error messages?

I would troubleshoot this by adding debugging statements in your script, such as write something to a file, or "sleep 10" so that you can check it's running via `ps`.

I have no better ideas for now :-(
Keep us posted.

alexrj, 2015-06-02:
@Hertafeld, do you have any update on this?

closed 2015-06-20 by alexrj