slic3r/Slic3r #2671: G4 commands inserted between some moves

SelfDestructor, 2015-02-18:
After I updated to 1.2.5 and recently 1.2.6 I have noticed my printer is halting quite a lot during printing and my LCD display flickers because of it. At first I thought my printer was about to die but after a lot of troubleshooting I finally decided to check the G-code.

Doing some quick searching through the entire G-code I think I have found that it  is at it's worst during short moves and looks like what's attached below.

If this is some sort of safety feature, is it possible to disable it somehow? If not currently, could you implement it in the next build?

Thanks.

> G1 X150.77 Y156.601 F5400
G4 P31
G1 X150.485 Y156.315 E58.91944 F4800
G4 P34
G1 X150.103 Y156.757 E58.94773
G4 P27
G1 X150.77 Y157.425 E58.99342
G4 P28
G1 X150.062 Y157.54 E59.02818
G4 P31
G1 X150.77 Y158.248 E59.0767
G4 P27
G1 X150.021 Y158.323 E59.11317
G4 P30
G1 X150.77 Y159.072 E59.16451
G4 P26
G1 X149.98 Y159.105 E59.20283
G4 P30
G1 X150.77 Y159.896 E59.25699
G1 X150.77 Y160.72 E59.29688
G4 P15
G1 X149.939 Y159.888 E59.35386
G4 P25
G1 X149.897 Y160.67 E59.39181
G4 P15
G1 X150.77 Y161.543 E59.45161
G1 X150.77 Y162.367 E59.4915
G4 P14
G1 X149.855 Y161.452 E59.5542
G4 P23
G1 X149.811 Y162.231 E59.59203
G4 P14
G1 X150.77 Y163.191 E59.65774
G1 X150.77 Y164.015 E59.69764
G4 P12
G1 X149.767 Y163.011 E59.76636
G4 P22
G1 X149.723 Y163.791 E59.80419
G4 P12
G1 X150.77 Y164.838 E59.87593
G1 X150.77 Y165.662 E59.91582
G4 P11
G1 X149.679 Y164.571 E59.99057
G4 P20
G1 X149.635 Y165.35 E60.0284
G4 P10
G1 X150.77 Y166.486 E60.10616
G1 X150.77 Y167.309 E60.14605
G4 P9
G1 X149.591 Y166.13 E60.22682
G4 P19
G1 X149.547 Y166.91 E60.26465
G4 P9
G1 X150.77 Y168.133 E60.34843
G1 X150.77 Y168.957 E60.38833
G4 P8
G1 X149.503 Y167.69 E60.47513
G4 P17
G1 X149.459 Y168.47 E60.51295
G4 P7
G1 X150.77 Y169.781 E60.60276
G1 X150.77 Y170.604 E60.64265
G4 P6
G1 X149.415 Y169.249 E60.73547
G4 P16
G1 X149.371 Y170.029 E60.7733
G4 P6

lordofhyphens, 2015-02-18:
G4 commands are used to slow print down if you have max. oscillation 
configured.

Using something smarter (like reducing feedrate) is a topic of interest, 
but it's a difficult problem with the way that slic3r is currently 
architectured.

SelfDestructor, 2015-02-18:
Really? I set the vibration limit many versions back and I cannot remember it ever doing it before.

Thank you so much!

closed 2015-02-18 by SelfDestructor