slic3r/Slic3r #2672: Can you add this problem to the milestone?

a4jp-com, 2015-02-18:
Can you add the following scrolling problem to the milestone? https://github.com/alexrj/Slic3r/issues/2521

alexrj, 2015-02-27:
I'd rather accepts patches, as my time is not infinite...

closed 2015-02-27 by alexrj

a4jp-com, 2015-02-28:
Thank you for looking at this problem. 

I just thought as my thread was made on January 10th (version 1.2.5) and it got pushed down the bottom almost no one would see it anymore. Also the the newest version 1.2.7-dev still has the same screen height model cutting problem. 

Sometimes it's just difficult to see if anyone has seen the threads in this site.

It would be great to have thread views on pages and thread views by admin on this site.

Or maybe the mark as read thing turned on.  

https://developer.github.com/v3/activity/notifications/#mark-as-read

Again thank you for you time and I understand you can't fix everything. It's just nice to list the bugs so they can maybe get fixed by this group or me when I get a bit better at programming in these new languages.