slic3r/Slic3r #2673: Feature Req:  Add option for labelling (sub)profiles, so that chosen profiles' labels can be added to output-filename-format

axs-gentoo, 2015-02-18:
I find myself needing to play with the output-filename-format a lot now due to the way the output filename format template is used solely for the "Send to printer" option.  I'm fine with this (ie I don't have a need specifically to manually name files each time), but it would be very helpful to me if I could have the selected filament profile as part of the filename.

alexrj, 2015-02-18:
https://github.com/alexrj/Slic3r/wiki/FAQ#how-can-i-specify-a-custom-filename-format-for-output-g-code-files

Does this work for you?

alexrj, 2015-02-18:
i.e. `[filament_preset]`

axs-gentoo, 2015-02-18:
Yes it does, thank you!!  I've been looking through the codebase for quite a while to try and find such an option.  I assume [print_preset] and [printer_preset] are also available?

closed 2015-02-18 by axs-gentoo

alexrj, 2015-02-18:
See the linked FAQ