slic3r/Slic3r #2674: freeglut3-dev needs to be installed

dwaq, 2015-02-20:
I'm following the install instructions to build the GUI from the latest source on Ubuntu and I get the following error messages:

    --> Working on Wx::GLCanvas
    Fetching http://www.cpan.org/authors/id/M/MB/MBARBON/Wx-GLCanvas-0.09.tar.gz ... OK
    Configuring Wx-GLCanvas-0.09 ... OK
    Building and testing Wx-GLCanvas-0.09 ... FAIL
    ! Installing Wx::GLCanvas failed. See /home/user/.cpanm/work/1424391688.15073/build.log for details.   Retry with --force to force install it.
    Don't worry, this module is optional.

This is the important bits from that build.log:

    Running Mkbootstrap for Wx::GLCanvas ()
    chmod 644 "GLCanvas.bs"
    "/usr/bin/perl" "/usr/local/share/perl/5.18.2/ExtUtils/xsubpp"  -noprototypes  -typemap         "/usr/share/perl/5.18/ExtUtils/typemap" -typemap "/usr/local/li$
    g++ -pthread   -c  -I. -I/usr/local/lib/perl/5.18.2/Wx -I/usr/lib/i386-linux-gnu/wx/include/gtk2-unicode-    release-2.8 -I/usr/include/wx-2.8  -D_REENTRAN$
    In file included from /usr/include/wx-2.8/wx/glcanvas.h:54:0,
                 from GLCanvas.xs:27:
    /usr/include/wx-2.8/wx/gtk/glcanvas.h:19:19: fatal error: GL/gl.h: No such file or directory
     #include <GL/gl.h>
                       ^
    compilation terminated.

I did some googling and found out that freeglut3-dev needs to be installed and that fixed the error. I'm not at all familiar with Perl, but it seems that this dependency should be added somewhere to get rid of this error.

axs-gentoo, 2015-02-20:
GL/gl.h is the most generic of the OpenGL headers.  Mesa, Nvidia-Drivers, and Ati-Drivers should all provide this (that is, a relevant -dev package related to them).  Freeglut-dev may also provide it but I would expect there is likely another dev package that provides the opengl headers too.

This dependency isn't really the fault of Slic3r, though; you should expect to need a -dev opengl package on your system if you plan to use an (optional) opengl-based perl package.  :)

gege2b, 2015-02-20:
Hi

Did you read https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux ?
 
It's said that freeglut3-dev should be installed if you want to use the 3D preview (anyway I don't know if there is a way to build the gui without 3D support)

dwaq, 2015-02-20:
@gege2b I did not see that page. Installing those dependencies fixed all of the problems I was seeing. It would be useful to update the Ubuntu section on <http://slic3r.org/download> to make that link more apparent. The extra paragraph in front of "See this page for instructions." is unnecessary since the Mac and Windows sections don't include it.

Besides that, I would consider this issue closed. Thanks for the help.

llluis, 2015-08-10:
@dwaq , can you please close this issue? Thanks

closed 2015-08-10 by dwaq