slic3r/Slic3r #2675: feature request-modular custom gcode (tool change, etc)

nicksears, 2015-02-21:
I love the new changes as of 1.2.6 with before/after layer change, but I'd really love to see before/after tool changes and unique for tool changes for that matter! 

That got me thinking, the best way to handle it might be to have no hardwired custom gcode. Instead we could have the same system as we have for extruders where you can add combinations of parameters specifying when the custom gcode is applied.

For instance before/after a print/layer/tool change for every nth layer, nth tool change, or change to a specific tool. 

Make my wish come true!

jonaskuehling, 2015-02-23:
I like that idea...

alexrj, 2015-02-23:
Not sure I understand fully...

nicksears, 2015-02-23:
Instead of preset custom gcode sections, have a button to add a new customizable section (like we have for the extruders).
For each section have a drop-down for if it's before/after and for a whole print/layer/tool and how often to call the g-code
screenshot mockup attached

nick
[image]

alexrj, 2015-02-27:
Nice mockup there. Thank you for the idea.