slic3r/Slic3r #2676: frequent crashes on 1.2.6 precompiled beta on win8x64

nicksears, 2015-02-21:
I've been using 1.2.6 on windows 8 x 64 for a while and noticed frequent crashes without explanation or dialogue. Usually mid editing part of a profile. Anyone else notice this.

cglusky, 2015-02-23:
Yes.  Both on OS X (latest) and Win7x64.  Had it crash this morning loading a small STL and then again editing a profile.  No errors or dialogues here as well.  Just closes.

nicksears, 2015-02-23:
I'm turning off incremental slicing to see if the actual slicing is causing
the close

cglusky, 2015-02-23:
good call.  so far turning off background processing in preferences has fixed it.  i'll update this as i play with it more.  but so far so good.  

makes me want a button in the UI to trigger a Preview and Layer update without needing to Export.

nicksears, 2015-02-23:
I've found it works as well so far, but I loved the new preview of layers
feature and it won't work without  the background processing.

alexrj, 2015-02-23:
Can you provide details and steps for reproducing this? I can't fix it if I can't reproduce it.

nicksears, 2015-02-24:
Windows 8.1x64 Slic3r 1.2.6x64 (1.2.6x32 seems to be ok?)
I've had it happen most commonly when messing with the custom gcode or
rotating/moving parts on the plater
Will post when I have a definite reproducible example

nicksears, 2015-02-24:
also, this happens whether or not I have any kind of graphics on screen (sometimes in the ptiner tab, sometimes in the plater changing profiles. Probably not a gfx issue then, right? The two systems I primarily use both have synchronized profiles, but one has nvidia, one has ati

nicksears, 2015-02-24:
dropbox link to folder with files I was using when I had it crash:
https://www.dropbox.com/sh/k93l0jwty4sml6s/AAAIDQuVouE7xk3j8-IkSz9fa?dl=0
just plate the stl's and load these profiles and move things around/change parameters.
again, I (think) I've only had it happen on x64

justmyopinion, 2015-02-24:
I see the same random crash now and then. It looks like some thread synchronization as an earlier incident repaired?. I can make it happen randomly just using rotate buttons during background updating. For some reason it does not happen with default settings (not yet).
W7/X64-1.2.6
[image]

cglusky, 2015-02-25:
just had my first crash after turning off background processing.  i was saving to gcode.  sorry, otherwise it's random.  much more stable with background processing turned off.  i was getting crashes during every use.  this is the first crash after processing several print jobs.

win7x64 SP1
intel i7
16GB RAM
AMD FirePro 5900
1TB HD

nicksears, 2015-02-25:
Anyone have an issue with x32 windows or the x32 on x64 windows?

nicksears, 2015-02-25:
Can confirm multiple crashes on 1.2.6 x32 and x64 using win8.1x64

Praesmeodymium, 2015-03-21:
Loading this thing into slic3r 1.26 crashes me with BG processing on
http://www.thingiverse.com/thing:455865

It also crashes when slicing somewhere around infilling layers
 my system
Operating System: Windows 7 Ultimate 64-bit
           Language: English (Regional Setting: English)
System Manufacturer: MSI
       System Model: MS-7721
               BIOS: BIOS Date: 10/23/12 14:57:22 Ver: 04.06.05
          Processor: AMD A4-5300 APU with Radeon(tm) HD Graphics     (2 CPUs), ~3.4GHz
             Memory: 8192MB RAM
Available OS Memory: 7644MB RAM
       DirectX Version: DirectX 11

nicksears, 2015-03-23:
that slices fine for me with 1.2.6 and 1.2.7 (x64) using my profiles  (https://www.dropbox.com/sh/f2zanfwk0ivwpsc/AAArdV2-hiugT9I7nxaEPBpna?dl=0)

ML3D, 2015-04-29:
I've also experienced crashes with the 64bit 1.2.6 Beta on Win 8.1. Slic3r just vanishes.
Happened a couple of times last night when I was adjusting layer height settings.

alexrj, 2015-06-02:
Okay, all these seem fixed in 1.2.7.  Feel free to reopen if not. Thank you!

closed 2015-06-02 by alexrj