slic3r/Slic3r #2677: Enabling Raft Causes Bad Z Layer Heights

DrAnonymous, 2015-02-21:
I have tried this for versions: 1.1.7 / 1.2.5beta / 1.2.6beta using OS X 10.10.1. I get the same behavior from repetier host invoking Slic3r or running Slic3r directly.

When I enable raft, I get incorrect Z layer heights, which make the model unprintable. Two examples, with a layer height of 0.2:

*1.2.6beta / Raft Height 1*
< Snip some gcode >
M109 S220 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
M106 S255
G1 E-1.00000 F1800.00000
G92 E0
**G1 Z0.200 F7800.000
! gcode for layers 0.2 / 0.4 / 0.6 are missing !
G1 Z0.800 F7800.000**
G1 X90.754 Y109.142 F7800.000
G1 E1.00000 F1800.00000
M106 S255
G1 X85.858 Y109.142 E1.25578 F3600.000

*1.2.6beta / Raft Height 3*
< snip some code >
M109 S220 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 E-1.00000 F1800.00000
G92 E0
**G1 Z0.200 F7800.000**
G1 X83.758 Y88.584 F7800.000
G1 E1.00000 F1800.00000
G1 X83.632 Y88.835 E1.00835 F1800.000
< finish layer and start next >
G1 X116.242 Y111.416 E63.57317
M106 S224.4
G1 E62.57317 F1800.00000
G92 E0
**G1 Z0.400 F7800.000**
G1 E1.00000 F1800.00000
G1 X116.368 Y111.164 E1.00835 F3600.000
< finish layer and start next >
G1 X83.758 Y88.584 E63.57317 F3600.000
M106 S91.8
G1 E62.57317 F1800.00000
G92 E0
**G1 Z0.600 F7800.000**
G1 E1.00000 F1800.00000
G1 X83.632 Y88.835 E1.00835 F3600.000
< finish layer and start next >
G1 X116.242 Y111.416 E63.57317
G1 E62.57317 F1800.00000
G92 E0
**G1 Z1.200 F7800.000
! This should have been a Z Height of 0.8 !**
G1 X111.977 Y109.142 F7800.000
G1 E1.00000 F1800.00000
M106 S255
G1 X85.858 Y109.142 E2.36456 F3600.000

Note how the layers properly go from 0.2 - 0.4 - 0.6 and then jump to 1.2. The settings are below.

; avoid_crossing_perimeters = 0
; bed_shape = 0x0,200x0,200x200,0x200
; bed_temperature = 110
; before_layer_gcode = 
; bridge_acceleration = 0
; bridge_fan_speed = 100
; brim_width = 0
; complete_objects = 0
; cooling = 1
; default_acceleration = 0
; disable_fan_first_layers = 1
; duplicate_distance = 6
; end_gcode = M104 S0 ; turn off temperature\nM84     ; disable motors\n
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 1
; fan_always_on = 0
; fan_below_layer_time = 60
; filament_diameter = 1.75
; first_layer_acceleration = 0
; first_layer_bed_temperature = 110
; first_layer_extrusion_width = 200%
; first_layer_speed = 30
; first_layer_temperature = 220
; gcode_arcs = 0
; gcode_comments = 0
; gcode_flavor = reprap
; infill_acceleration = 0
; infill_first = 0
; layer_gcode = 
; max_fan_speed = 100
; min_fan_speed = 35
; min_print_speed = 10
; min_skirt_length = 0
; notes = 
; nozzle_diameter = 0.4
; only_retract_when_crossing_perimeters = 1
; ooze_prevention = 0
; output_filename_format = [input_filename_base].gcode
; perimeter_acceleration = 0
; post_process = 
; pressure_advance = 0
; resolution = 0
; retract_before_travel = 2
; retract_layer_change = 1
; retract_length = 1
; retract_length_toolchange = 10
; retract_lift = 0
; retract_restart_extra = 0
; retract_restart_extra_toolchange = 0
; retract_speed = 30
; skirt_distance = 6
; skirt_height = 1
; skirts = 0
; slowdown_below_layer_time = 30
; spiral_vase = 0
; standby_temperature_delta = -5
; start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
; temperature = 220
; threads = 2
; toolchange_gcode = 
; travel_speed = 130
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 0
; z_offset = 0
; dont_support_bridges = 1
; extrusion_width = 0
; first_layer_height = 0.2
; infill_only_where_needed = 0
; interface_shells = 0
; layer_height = 0.2
; raft_layers = 3
; seam_position = random
; support_material = 0
; support_material_angle = 0
; support_material_contact_distance = 0.2
; support_material_enforce_layers = 0
; support_material_extruder = 1
; support_material_extrusion_width = 0
; support_material_interface_extruder = 1
; support_material_interface_layers = 3
; support_material_interface_spacing = 0
; support_material_interface_speed = 100%
; support_material_pattern = pillars
; support_material_spacing = 2.5
; support_material_speed = 60
; support_material_threshold = 0
; xy_size_compensation = 0
; bottom_solid_layers = 3
; bridge_flow_ratio = 1
; bridge_speed = 60
; external_fill_pattern = rectilinear
; external_perimeter_extrusion_width = 0
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_perimeters = 1
; fill_angle = 45
; fill_density = 20%
; fill_pattern = honeycomb
; gap_fill_speed = 20
; infill_every_layers = 1
; infill_extruder = 1
; infill_extrusion_width = 0
; infill_overlap = 15%
; infill_speed = 80
; overhangs = 1
; perimeter_extruder = 1
; perimeter_extrusion_width = 0
; perimeter_speed = 60
; perimeters = 3
; small_perimeter_speed = 15
; solid_infill_below_area = 70
; solid_infill_every_layers = 0
; solid_infill_extruder = 1
; solid_infill_extrusion_width = 0
; solid_infill_speed = 20
; thin_walls = 1
; top_infill_extrusion_width = 110%
; top_solid_infill_speed = 15
; top_solid_layers = 3

alexrj, 2015-03-06:
Have you actually tried printing this? It's not wrong, that's by design.  A 0.4mm nozzle will extrude beads having a 0.4mm diameter. The default *Support material contact distance* value is 0.2mm, so the first object layer will be at +0.6mm from raft.

You might want to adjust the *Support material contact distance* option if you get bad adhesion between object and raft.

Background of the distance logic is here: http://hydraraptor.blogspot.it/2012/08/peel-able-support.html

closed 2015-03-06 by alexrj

DrAnonymous, 2015-03-06:
Yes, I tried multiple times to get a print to work and always ended up with
a mess. I have gotten it to work before, which makes me think that some
other setting(s) were causing the issue. I do understand the article you
referred to, thank you for that link. I still believe something is wrong if
you study my samples.

In my first example, I did NOT remove any gcode for z height of 0.2, I
simply put a comment in to show gcode was never generated. This means
the first extrusion occurs at a bed height of 0.8mm. It doesn't matter if
the 0.8 layer is support or the model, because that far above any material
with a 0.4 head will never stick. Why would the gcode show setting Z height
and then not extrude or move at all, before going to a new z height?

The second example I showed DOES have gcode for three z heights, but it
also moves from 0.8 to 1.2. If I understood the article you linked, it
would skip 1.0 for the contact distance and then start extruding at the 1.2
layer which does make sense.

I believe there are two problems:

- The first example shows incorrect gcode generation
- The second should work, but something else is preventing the print from
working (speed setting maybe?)