slic3r/Slic3r #2679: 1.2.6-dev: Expected hole in simple cylinder has bridging strands in center

lordofhyphens, 2015-02-21:
Title says it all, seeing lines in the output gcode (screencap is from slic3r) on 1.2.6-dev. The print does look like the gcode preview. Also,  there's a 1-line gap on the bottom layer which I don't understand, which is visible in the gcode preview and was in the print.

[image]

STL, configuration, and resultant gcode:
http://www.2shared.com/file/QAW9P3I5/weird-infill.html
Remember to use the second link (first is an advertisement).

OS is Debian/Linux 8 (Jessie)

alexrj, 2015-02-22:
At least we know that G-code preview is realistic ;-)

lordofhyphens, 2015-02-22:
Yes, but that is coming from an incorrectly generated tool path :p

alexrj, 2015-02-22:
Okay, 2655f3f816b07abcb6f3df9b633dcbaec259cac7 fixes this.

closed 2015-02-22 by alexrj