slic3r/Slic3r #2684: [print_preset] no longer available

lordofhyphens, 2015-02-22:


alexrj, 2015-02-23:
Uh? It works for me…
How to reproduce?

alexrj, 2015-03-06:
@lordofhyphens, any news? :)

lordofhyphens, 2015-03-09:
The problem seems to be that it is not updating the variable. If I open a
new instance of slic3r the variable isn't available. If I export gcode and
then switch profiles, the name used is the previously active profile.

nicksears, 2015-03-11:
I had an issue where all of my preset placeholders just showed up as their text "print_preset", etc.
Deleting  slic3r.ini (renaming to *.bak) made slic3r start like it was the first time, but fixed the issue.
Had to do a first startup and select expert mode, but worked fine (doesn't delete your presets, but backup just in case)

alexrj, 2015-06-02:
@lordofhyphens, I tested with 1.2.8-dev now and it looks fixed.
Can you confirm? Thank you

alexrj, 2015-06-21:
@lordofhyphens I'm not able to reproduce this with 1.2.9 as well. I tried setting the Output Filename Format to `[input_filename_base]_[print_preset].stl`, then I loaded a file into the plater and then I clicked on *Export G-code...*. The save dialog has the full expanded name with the correct print preset name. Can you help checking please?

lordofhyphens, 2015-06-21:
I will look into reproduction; take a video or something. It doesn't seem
to have a real effect on the output.

alexrj, 2015-06-21:
Well, unless this can be reproduced I'm closing it...

closed 2015-06-21 by alexrj

nicksears, 2015-06-21:
The issue is that the export dialog loses the actual preset names and just shows the [print_preset], etc.
I have yet to reproduce since I'm using later versions, but it seems to be an issue with the slic3r.ini file in the profiles folder. If anyone ever reproduces it save the slic3r.ini file, I think it may have had something to do with the background processing (which was pretty buggy before 1.2.7). Now it seems great.