slic3r/Slic3r #2686: No anchor for edges of interior angle (missing internal perimeter?)

markwal, 2015-02-23:
I'm running 1.2.7-dev

The layer immediately below:
[image]

The layer in question:
[image]

Note that the solid infill layer for the inset over every other honeycomb is unsupported/unanchored.

Or perhaps it should decide to do a solid infill layer here?

stl is here: http://www.thingiverse.com/download:1087135  It is the ILoveYouBoxLid.stl part of http://www.thingiverse.com/thing:676630/#files

alexrj, 2015-02-23:
I don't understand. It looks very okay. The square solid infill is laid over sparse infill. What's the problem? Do you have a picture of the printed object showing the problem?

markwal, 2015-02-24:
The loops over the infill droop or peel up.  But I see what you mean, as it is laid down it should anchor to the adjacent.  Probably this is fine, I need to recalibrate.

closed 2015-02-24 by markwal