slic3r/Slic3r #2687: 3D view not rendering part

fotis400, 2015-02-23:
I just pulled latest changes, commit e88cf466fedb19ca51ba55db7d82dfbb9b3a3198. I built from sources, and I didn't face any compilation errors.

I am using Fedora 21 x86_64.

While models are loaded correctly and displayed correctly in 2D view, they are not showing up at all in the 3D view. 
[image]

alexrj, 2015-02-23:
Were you able to see the 3D models before the git pull?
How large is the model you loaded?
Does the print bed display before you load any model (thus while empty)?

fotis400, 2015-02-23:
Yes, before the pull it was working correctly. Last time I pulled was about 1-1.5 week ago, so it must be one of the latest commits.

I tried with many models, even very small, and very large ones.

No. The print bed is not rendering at all, even just after I start Slic3r. It is interesting that when I select one of the models, the view is moving (in respect to axis vectors), and zooming to were the model should be, but no bed, or models.

axs-gentoo, 2015-02-24:
This is working fine for me, on Gentoo Linux, as of commit 45c91b2ae93 ...  whatever the issue is, it's either already been fixed or it's likely something Fedora-specific (dependencies, maybe?)

fotis400, 2015-02-24:
I tested again with 45c91b2ae93ca89ca837e122c76f20d2458e1bcd .This time I was a bit more cautious at the build process and there is an error... It points me to this log file:

http://pastebin.com/ddqTSYem

Also while running Slic3r I get these errors:

libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast

axs-gentoo, 2015-02-24:
Ah, you're missing freeglut or some other GLUT implementation.  Other possible issues with your 3d setup there too..

fotis400, 2015-02-24:
I just checked. I have installed both freeglut and freeglut-devel...
Any other clue?

axs-gentoo, 2015-02-24:
Well, the build error in that log is on the perl OpenGL module, and it's not finding GL/glut.h (which should be in /usr/include )..  you'll need to figure out why.  

I don't know if this is pertinent or not, but on Gentoo Linux, I skip the main 'Build.PL' and Cpanminus stuff completely, and just do a 'perl Build.PL' from within the XS dir to build that library.  You'll need to have all of the system libs installed though (perl's OpenGL, ExtUtils::ParseXS, Wx, Alien-wxWidgets, WxOpenGL, Math::PlanePath, etc etc) in order for slic3r to actually work.

fotis400, 2015-02-25:
The GL/glut.h file is there as it has to be. I just tried Build.PL from XS but no luck.
All the mentioned libraries are installed. 
Still no luck...

Keep in mind that Slic3r was working correctly. I haven't changed anything to my system, neither updated it, I just pulled Slic3r and the problem appeared. It must be something related to the last week's changes...

Thank you very much for your time....

gege2b, 2015-02-25:
do you use 

sudo perl Build.PL 

or 

perl Build.PL --sudo ?

fotis400, 2015-02-25:
from the dir xs/ I can only sudo perl Build.PL, the other one just gives me permission errors.

from the parent dir I just tried both, but the problem persists...

alexrj, 2015-02-25:
@fotis400, you only have to run the Build.PL script in the main directory, not the other one.   Does the execution of it fail anyhow? Do you see any warning or error message in its output?

fotis400, 2015-02-25:
yes, normally I use only the Build.PL in the main directory.
Please check the outputs

sudo perl Build.PL: (seems ok to me)
http://pastebin.com/ww8FZHur

sudo perl Build.PL --gui:
http://pastebin.com/0sC2TGPx

./slic3r.pl
http://pastebin.com/UnwApnS5

axs-gentoo, 2015-02-25:
The thing about using the main Build.PL is that it's not -just- building the updated slic3r code, it's also going to use App:Cpanminus to upgrade all of the dependencies.  If you see the "perl Build.PL --gui" log, it's failing to install the perl OpenGL module.  WX::GLCanvas doesn't work without OpenGL, as far as I've been able to determine here.

You can try installing a system copy of the perl OpenGL module via a fedora package (if there is one), it's possible that will make things 'just work'.  Otherwise though you will need to sort out the issue that's keeping the OpenGL module from building via 'perl Build.PL --gui'.

Regarding the libGL error -- are you bychance using nvidia-drivers for X11 ?  If so this might be part of the issue.  Unfortunately it's a bit beyond me how to sort out libGL issues between mesa vs  proprietary drivers, but it's a place to start looking.

fotis400, 2015-02-25:
I have all the perl-OpenGL related packages installed. Unfortunately I am using the proprietary driver, and I cannot change this.

What is most important for me, is not just making it work for me. It is that is was working, and only one of the latest commits broke it. I am insisting for checking the code, to make Slic3r bug-free, not only for me but for anyone.

Tomorrow I will test all latest commits, to see which one causes the problem.

gege2b, 2015-02-26:
can you try to make a 

    xhost + && sudo perl Build.PL --gui && xhost -

this authorize X to allow connection from any user, even root, before building, and then sets all the stuff back

I found that compiling OpenGL with root/sudo privileges fails (because of root connection to X is not authorized), so I always use this and I got no problem at all

lordofhyphens, 2017-03-19:
Can't reproduce. @fotis400 if you are still having issues let me know and I'll reopen.

lordofhyphens, 2017-03-19:
Note: You do not build Slic3r with root anymore.

closed 2017-03-19 by lordofhyphens