slic3r/Slic3r #2688: No close buttons for pop-ups

fotis400, 2015-02-23:
I have this problem for months now, but for some reason I never posted the issue...

I am using Fedora 21 x86_64, (but also previously tested in Fedora 20), with Gnome 3.14.

Each time a pop-up window opens, there is no bar on top of it, so there is no close button. The only way to close the window it by pressing [Alt]+[F4]. Previously it was rather annoying, (as 3D view was a separate window), now not so much, but it is still a bug.

Take a look at the "about" window.
[image]

axs-gentoo, 2015-02-24:
What version of wxGTK and Alien-wxWidgets are you using, in Fedora 21?  the lack of window decorations/controls would seem to be an issue between those two packages rather than (i expect) something specific to Slic3r.  Do you see this issue with any other package that renders windows with the help of wxGTK?

fotis400, 2015-02-24:
Well I can't tell for sure.

Fedora is naming differently wx versions than other distros. There are separate packages for version 3 and above, and separate for below 3.

Currently I have installed wxGTK (v.2.8.12-13fc21) and wxGTK3 (v.3.0.2-2.fc21), also I have installed perl-Alien-wxWidgets (v.0.51-13fc21)

I can't tell which one is being used. Maybe version v.2.8 as the package is named as in other distros... How can I find it, and how can change which one is used? (I'm building from sources).

I haven't faced any problems before with wx.

axs-gentoo, 2015-02-24:
Alien-wxWidgets 0.64 supports no newer than wxGTK-2.9, so I expect 0.51 would only go as new as 2.8.  You need 0.65 plus a couple of patches to support wxGTK-3, and it's possible you might need wxGTK3 to have proper window decorations/controls in Gnome-3.14 ...

axs-gentoo, 2015-02-25:
OK, so on Gentoo with Gnome-3.14, using Alien-wxWidgets 0.64 and wxGTK-2.8, apparently we do see a title bar, but no buttons; have to right-click on the titlebar and then choose close from the drop-down.  So there is possibly an issue here either with older wxGTK on Ghome-3.14, or the way the dialog windows/widgets are being created within slic3r that's not entirely Gnome-3.14 optimal.

OR, it's a limitation in gnome-3 itself.  You might need this, or something similar:  https://github.com/danielkza/Gnome-Shell-Window-Buttons-Extension

tomgeor, 2015-02-26:
I have the same issue on my Fedora 21 workstation.  I'm using Cinnamon rather than Gnome 3 proper.

alexrj, 2015-03-21:
This is weird. On Ubuntu 12.04 I see both title bar and close button...

alexrj, 2015-03-28:
@fotis400, can you please run Slic3r from command line with the `--debug` option? It will tell the wxWidgets and the Wx versions.

fotis400, 2015-03-29:
Here's the output:
http://pastebin.com/Nmn2rsUF

alexrj, 2015-05-04:
Relevant link: https://forums.wxwidgets.org/viewtopic.php?t=35312

Mikk36, 2015-05-25:
This isn't specific to just Linux, also happens on my Windows build with latest git commit.

alexrj, 2015-06-02:
Okay, I added a close button and support for Esc key and it also closes when you click inside the window. :-)

closed 2015-06-02 by alexrj