slic3r/Slic3r #2690: Modifier is generating undesired internal perimeters

thiagopeixoto16, 2015-02-24:
The image shows what I mean by undesired internal perimeters:
[image]
It is there only for few layers, not all.

Here find below the STLs, config and generated G-code:
https://www.dropbox.com/sh/cpec6t0ktotff4q/AADTZXPxuxx6YTx5kzdbCj09a?dl=0

Vicious-one, 2015-02-27:
That's actually a feature, very useful in some situations. 

If you don't override perimeter-related parameters with a modifier (width, number of perimeters, etc), it should not generate additional perimeters.

If you want to do it anyway, make your modifier mesh fit inside the outer shell.

alexrj, 2015-02-27:
Yes, as @Vicious-one says the separate perimeters are generated only if the two regions have different perimeter settings. If they have the same settings, a common set of loops is generated.

closed 2015-02-27 by alexrj

thiagopeixoto16, 2015-02-27:
I got it and noted! Thanks!