slic3r/Slic3r #2691: Hot end impacts previously built object

c-born, 2015-02-25:
Win7/64, 1.2.7-dev  current head 45c91b2ae93ca89ca837e122c76f20d2458e1bcd has taken on a nasty turn when printing sequential objects. After completing one object it moves to the next, then comes back to the point it left the first and crushes it. I thought at first it may be in the retract code, but I disabled retract and it still happened.

In the picture it starts on the object at lower right, completes it, moves to the upper object, then returns to its departure point on the first object and drives Z down to 0.2. It does the same on completing the top object, making a mess of the object and the hot-end.

In the GCode you can see the offending line at 1697.
At line 1690 it finished the first object and moves into position for the second.
It then waits for temperature, moves back to the first object and at line 1699 drives down into it, destroying it.

[image]

Settings at  https://dl.dropboxusercontent.com/u/7955521/config1.ini

a4jp-com, 2015-02-25:
Can you include the 3D objects as well for testing?

c-born, 2015-02-25:
It isn't dependent on the objects - here's the same result with two standard 0.5mm-thin-wall.stl test parts, you can see the problem at lines 2284 and 2285.

[image]

alexrj, 2015-03-06:
Thank you @c-born, I pushed a fix. Can you confirm the issue is gone?

c-born, 2015-03-06:
@alexrj,
I just tested with current head 9332c2179166e93ab2f2fca0a1fa9099f845699c and the two thin-wall test parts.

The issue seems to be still there, also it now seems to be generating a weird diagonal across the part and a lot of extra movements.

[image]

I can't be sure my build is good. It passed the tests, but failed to install "threads", as per build output:

```
D:\usr\CNC\Slic3r>perl Build.PL
App::cpanminus is up to date. (1.7027)
Class::XSAccessor is up to date. (1.19)
Encode::Locale is up to date. (1.04)
ExtUtils::MakeMaker is up to date. (7.04)
ExtUtils::ParseXS is up to date. (3.24)
skipping R/RJ/RJBS/perl-5.20.0.tar.gz
File::Spec is up to date. (3.47)
Getopt::Long is up to date. (2.45)
IO::Scalar is up to date. (2.110)
Math::PlanePath is up to date. (118)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.000000)
skipping R/RJ/RJBS/perl-5.18.4.tar.gz
Scalar::Util is up to date. (1.41)
Test::Harness is up to date. (3.35)
Test::More is up to date. (1.001014)
Thread::Semaphore is up to date. (2.12)
Time::HiRes is up to date. (1.9726)
Unknown option: configure-args
--> Working on XML::SAX::ExpatXS
Fetching http://www.cpan.org/authors/id/P/PC/PCIMPRICH/XML-SAX-ExpatXS-1.33.tar.gz ... OK
Configuring XML-SAX-ExpatXS-1.33 ... OK
Building and testing XML-SAX-ExpatXS-1.33 ... FAIL
! Installing XML::SAX::ExpatXS failed. See C:\Users\cborn\.cpanm\build.log for details.
Don't worry, this module is optional.
--> Working on threads
Fetching http://www.cpan.org/authors/id/J/JD/JDHEDDEN/threads-1.98.tar.gz ... OK
Configuring threads-1.98 ... OK
Building and testing threads-1.98 ... FAIL
! Installing threads failed. See C:\Users\cborn\.cpanm\build.log for details.
Cleaning up build files
Cleaning up configuration files
cpanm (App::cpanminus) 1.6103 on perl 5.014002 built for MSWin32-x64-multi-thread
Work directory is C:\Users\cborn/.cpanm/work/1425639423.12176
You have make C:\dev\CitrusPerl\mingw64\bin\dmake.exe
You have LWP 6.08
Falling back to Archive::Tar 1.76
--> Working on ./xs
Entering D:/usr/CNC/Slic3r/xs
Configuring D:/usr/CNC/Slic3r/xs ... Running Build.PL
'cc' is not recognized as an internal or external command,
operable program or batch file.
Created MYMETA.yml and MYMETA.json
Creating new 'Build' script for 'Slic3r-XS' version '0.01'
OK
Checking dependencies from MYMETA.json ...
Checking if you have Test::More 0 ... Yes (1.001014)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.24)
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.24)
Checking if you have ExtUtils::XSpp 0.17 ... Yes (0.1700)
Checking if you have Module::Build 0.3601 ... Yes (0.40)
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
gcc -c -s -O2 -DWIN32 -DWIN64 -DCONSERVATIVE  -DPERL_RELOCATABLE_INCPUSH -DPERL_TEXTMODE_SCRIPTS -DUSE_SITECUSTOMIZE -DPERL_IMPLICIT
_CONTEXT -DPERL_IMPLICIT_SYS -fno-strict-aliasing -mms-bitfields -xc++ -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -Isrc -Ibuil
dtmp -s -O2 -I"C:\dev\CitrusPerl\CitrusPerl\lib\CORE" -I"C:\dev\CitrusPerl\mingw64\x86_64-w64-mingw32\include" -o "src\libslic3r\Lin
[... 198 lines cut ...]
t\07_extrusionpath.t ............ ok
t\08_extrusionloop.t ............ ok
t\09_polyline.t ................. ok
t\10_line.t ..................... ok
t\11_clipper.t .................. ok
t\12_extrusionpathcollection.t .. ok
t\13_polylinecollection.t ....... ok
t\14_geometry.t ................. ok
t\15_config.t ................... ok
t\16_flow.t ..................... ok
t\17_boundingbox.t .............. ok
t\18_motionplanner.t ............ ok
t\19_model.t .................... ok
t\20_print.t .................... ok
All tests successful.
Files=19, Tests=454,  2 wallclock secs ( 0.13 usr +  0.02 sys =  0.14 CPU)
Result: PASS
Building Slic3r-XS
Files found in blib\arch: installing files in blib\lib into architecture dependent library tree
Installing C:\dev\CitrusPerl\CitrusPerl\site\lib\auto\Slic3r\XS\XS.dll
OK
Successfully installed Slic3r-XS-0.01
1 distribution installed
The following prerequisites failed to install: threads
```

alexrj, 2015-03-22:
Okay, this is caused by avoid_crossing_perimeters trying to be smarter than needed with sequential printing.

alexrj, 2015-03-23:
Fixed, thank you.

closed 2015-03-23 by alexrj