slic3r/Slic3r #2692: 1.2.6 - Parts not auto arranged

quillford, 2015-02-25:
If I select multiple files to add with the add button, it will not arrange them, and the parts will be stacked on top of each other on the center.

Version: 1.2.6 pre compiled
OS: OS X 10.10.3
[image]
[image]

alexrj, 2015-02-27:
They are actually autoarranged after loading, but the arrange algorithm is not too smart so sometimes it prefers to skip arranging. Can you try to repeat the same procedure as above, and then click the Arrange button manually? what happens?

quillford, 2015-02-27:
It works if I click arrange after importing. However I just realized that I have auto centering parts turned off in my settings. Would that affect it?

alexrj, 2015-03-06:
I found my own comment in code saying that auto arrange is intentionally not performed when autocentering is disabled. I didn't remember that. The reason is that if user has positioned some objects in custom positions, having all of them rearranged upon loading a new object would most likely disappoint them…

czyz, 2016-01-31:
I recall this feature working in the past in a way that was useful -- dragging in a series of parts would distribute them on the bed with no overlaps but clustered around the center. As of Slic3r 1.2.9 on OS X some of this behavior seems to have changed for the worse. 

If I drag in a set of several separate STL files, and 'Auto-center parts' is unchecked, all the parts are placed in the center of the bed, completely overlapping one another. Subsequent hits of the 'Arrange' button do nothing. If I move one part and hit the 'Arrange' button, the group consisting of the overlapped parts and the single separated part are centered in the bed.

If I drag in a set of separate STL files, and 'Auto-center parts' is checked, the parts are distributed around the center of the bed. But surprisingly, there's still lots of overlap between parts. Here's what it looks like when I drag in the files for [this device](https://www.kickstarter.com/projects/1064868871/smoke-signal-a-safety-system-for-3d-printers) (which include rafts as part of the objects), this happens:
[image]

In an ideal world the software could automatically distribute objects on the bed such that they don't overlap. I think this used to happen by default, but maybe my recollection is fuzzy.

alexrj, 2017-03-18:
Behavior was changed now - parts are always arranged.

closed 2017-03-18 by alexrj