slic3r/Slic3r #2693: Honeycomb infill is offset - 1.2.6

BJoey, 2015-02-25:
didn't find any similar issue so i'll open one:
since using Slic3r 1.2.6, the honeycomb infill layers do not always align correctly, as seen in the picture. Does not connect with perimeters as well and seems to get worse with smaller layer heights. this causes a great loss of stability and top layers are not properly filled due to the lack of material in the layers below.
Did I change any option to make this happen or is it a real bug?
[image]

a4jp-com, 2015-02-25:
I have the same problem.

alexrj, 2015-02-27:
@BJoey, see the guidelines. We need to reproduce your problem in order to help or take any action.

a4jp-com, 2015-02-28:
If BJoey can't post a file I can try and make a 3D model again with this problem. I deleted all the files I had before because they just looked ugly. I just had a land model with a honeycomb infill set at about 20 or 30 percent. Kind of a box with the bottom turned off.

BJoey, 2015-02-28:
Hey sry I can't figure out how to upload a file, but that proplem does not depend on a specific STL file. The picture above is a print of this one: http://www.thingiverse.com/thing:65168
I can post the GCode of one of those prints or Slic3r settings if it helps:

# generated by Slic3r 1.2.6 on Sat Feb 28 11:09:23 2015
avoid_crossing_perimeters = 0
bottom_solid_layers = 4
bridge_acceleration = 0
bridge_flow_ratio = 1
bridge_speed = 30
brim_width = 0
complete_objects = 0
default_acceleration = 0
dont_support_bridges = 0
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 1
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0
fill_angle = 45
fill_density = 50%
fill_pattern = honeycomb
first_layer_acceleration = 0
first_layer_extrusion_width = 200%
first_layer_height = 0.3
first_layer_speed = 50%
gap_fill_speed = 20
gcode_comments = 0
infill_acceleration = 0
infill_every_layers = 2
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 15%
infill_speed = 60
interface_shells = 0
layer_height = 0.1
min_skirt_length = 0
notes = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 40
perimeters = 3
post_process = 
raft_layers = 0
resolution = 0
seam_position = nearest
skirt_distance = 3
skirt_height = 1
skirts = 1
small_perimeter_speed = 30
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 60
spiral_vase = 0
standby_temperature_delta = -5
support_material = 0
support_material_angle = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_extruder = 1
support_material_interface_layers = 0
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 2.5
support_material_speed = 60
support_material_threshold = 0
thin_walls = 1
threads = 2
top_infill_extrusion_width = 0
top_solid_infill_speed = 40
top_solid_layers = 4
travel_speed = 130
xy_size_compensation = 0

lordofhyphens, 2016-07-12:
Looks like it's related to the width detection or infill_every_layers. For some reason, the first layer of internal infill is getting a wider infill than the ones above it and it's probably messing with the spacing.

first layer of infill:
[image]
second layer of infill
[image]

Only the lowest layer is getting the declared width of 1.98mm (in the gcode).

Setting the width manually sorts things out, as did setting infill_every_layers = 1 or PR #3410. 
@BJoey you could use my branch or do what I mentioned above. I would suggest using my branch because  combined infill isn't going to work like you expect it to anyway with the current code (it will make the part weaker).

Branch with fix:
https://github.com/lordofhyphens/slic3r/tree/fix-combine-infill

alexrj, 2016-12-21:
@lordofhyphens, since your PR was merged I assume this issue is fixed. Is this right?

lordofhyphens, 2016-12-21:
I'll check all of these PRs tonight when I get home from work and close
them if my test cases pass.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

lordofhyphens, 2016-12-22:
It looks like they're aligning properly now according to the provided test case, although the preview is drawing the upper layers thinner (this may just be because of combine-infill not having a good layer-by-layer draw method). I'd appreciate it if @BJoey would check it as well on his end :)

alexrj, 2016-12-22:
@lordofhyphens, rendering of combined infill should be proper. If last layer is thinner maybe it's because an even number of layers are configured for combination and there is an odd number of them, so the last one is not combined... is this the case?

lordofhyphens, 2016-12-24:
0.90 (first layer is 0.3, successive are 0.2)
[image] 
It's the bottom sparse infill layer that's fat.

0.70 [image]

4.90 - last layer before the sloping begins 
[image] 

Last layer of combined sparse infill is at 19.70, so I'm not seeing an odd/even thing going on here.

lordofhyphens, 2017-03-11:
[image]
Current behavior looks like it's tied to auto-width. The upper layers are drawn at a higher effective layer height than the lowest layer (and it's known that the auto-calculation has surprising results at low layer heights.

lordofhyphens, 2017-03-11:
[image]

lordofhyphens, 2017-03-11:
0.5 fixed width for infill:
[image]

lordofhyphens, 2017-03-11:
I think it's behaving as it's been designed to, though, so I think we can close this and I'll reference this in either another issue for automatic line widths (or to another one if it exists)

closed 2017-03-11 by lordofhyphens