slic3r/Slic3r #2696: Random crash in 1.2.6 on the Mac

Astroman, 2015-02-27:
Slic3r 1.2.6 running on my MacBook Pro OS X 10.10.2 randomly crashes after running for a while. Unfortunately, I haven't been able to pinpoint when or why this is happening. 1.2 did not have this problem. Let me know if I can provide more info.

alexrj, 2015-02-27:
We'd need everything that can help reproducing it.  Config files, STL files, instructions...

Astroman, 2015-02-27:
OK, next time it crashes, I’ll gather up the files.

MikeFitz73, 2015-03-01:
this has also been happening to me on windows xppro, slic3r 1.2.6.  I notice it happens if I am going back between 3d preview window and printer settings a lot.

bardiir, 2015-04-15:
I got these crashes too. It crashes sometime after generating the brim. It crashes for quite a wide range of objects but pretty much non-reproductive. Once it crashes it will crash always until some later time when the same settings do work perfectly fine. I don't know what actually changes except for the applications running in parallel.

What I could reproduce however is, it never crashes on quick slice. So it probably has something to do with the UI / Showing the sliced part on the plater or the Layers. 

GCode generation with the same settings/models on the cli seems to work fine too, but slicer ignores my bed size settings and prints with a 0,0 origin thus way off the print bed on my printer, I'll look into that bug later on when I got time to see if I can reproduce and why it might be happening (I'm loading multiple configs, maybe loading only one combined config works or something like that).

When the crash is happening I get this in the Console:

    15.04.15 07:01:21,381 slic3r[1095]: WARNING: The Gestalt selector gestaltSystemVersion is returning 10.9.2 instead of 10.10.2. Use NSProcessInfo's operatingSystemVersion property to get correct system version number.
    Call location:
    15.04.15 07:01:21,391 slic3r[1095]: 0   CarbonCore                          0x900897c7 ___Gestalt_SystemVersion_block_invoke + 135
    15.04.15 07:01:21,391 slic3r[1095]: 1   libdispatch.dylib                   0x93f23130 _dispatch_client_callout + 50
    15.04.15 07:01:21,391 slic3r[1095]: 2   libdispatch.dylib                   0x93f230b5 dispatch_once_f + 251
    15.04.15 07:01:21,392 slic3r[1095]: 3   libdispatch.dylib                   0x93f240d8 dispatch_once + 31
    15.04.15 07:01:21,392 slic3r[1095]: 4   CarbonCore                          0x9001be88 _Gestalt_SystemVersion + 1050
    15.04.15 07:01:21,392 slic3r[1095]: 5   CarbonCore                          0x9001ba39 Gestalt + 150
    15.04.15 07:01:21,392 slic3r[1095]: 6   libwx_baseu-3.0.0.0.0.dylib         0x0603b12d _Z19UMAGetSystemVersionv + 45

alexrj, 2015-06-02:
Can you please try 1.2.7 and tell whether it's more stable for you?
Thank you.

bardiir, 2015-06-02:
Haven't had a crash with 1.2.7 yet. I've been using that since 3-4 days now. But it might also be a side effect. I've noticed the crashes seem to occur mostly when you add a 3D model and then (while it's processing) you spin the platform already while beeing switched into preview mode. I've done a lot of complex slicings in the last week or so, so there is not a lot of dragging happening during the slicing as it takes a while anyway and you don't get impatient playing with the platform as you switch away anyway on long slicings :D

alexrj, 2015-06-02:
Heh :-)
Then I guess we can close it. Feel free to report if you find anything wrong.

closed 2015-06-02 by alexrj