slic3r/Slic3r #2697: Custom infill extrusion width causes infill to not reach perimeters

hhhudson, 2015-02-28:
Slic3r version: 1.2.6 experimental

Operating system: OSX 10.10.1

Steps to reproduce the issue:

* Set infill extrusion width to a small number (0.1, for example)
* Slice a model with a flat upper surface
* (Optional) Print

Expected result:

Infill will reach perimeter

Actual result:

Infill area appears to be calculated using the standard extrusion width, and does not reach perimeter. This is visible in the preview pane and on the actual print.

[image]

[image]

G-code:

https://gist.github.com/hhhudson/ed7ad080f56415a68f55

Config.ini:

https://gist.github.com/hhhudson/d25025ba57d20e7a426e

STL, OBJ or AMF input file:

http://www.thingiverse.com/thing:39050

alexrj, 2015-03-06:
Thank you for the very detailed report, I appreciate it.

alexrj, 2015-03-06:
This can also be reproduced with default settings and a low Top Solid Infill Extrusion Width (like 0.1mm).
Note to self: this is caused by the `$distance_between_surfaces` logic in Fill.pm

I'll fix this in 1.3.x as it requires some important changes to the code.

lordofhyphens, 2017-03-06:
How's this relate to the move to c++ infill?

alexrj, 2017-03-11:
I'm not able to reproduce this on master anymore, but I tested again on 1.2.6 and I can reproduce with that one. The issue is fixed but I need to write a regression test before closing this.

alexrj, 2017-03-12:
Regression test added. We can safely close this issue. Thank you for reporting it.

closed 2017-03-12 by alexrj